opentx/opentx #3828: Total Flight Time Horus

Bumblebee78, 2016-09-18:
Enhancement: I would like a timer, which run from a preset time up. For total flight time. Now you can set it only to 1:00:00 and the timer runs down.

MyRCode, 2016-09-19:
If you set the timer to 0:00:00 it will count up.  I use one timer for Total flight time per model.

3djc, 2016-09-19:
You also need to set it persistant

Bumblebee78, 2016-09-19:
I know, if the timer start by 0:0:0 it count up.   I have a yak on the taranis whit total time 21:33:00.  i would like start the count by this time.

Bug:
over 00:60:00 the next time is not 01:01:00 its 00:61:00.

3djc, 2016-09-19:
That bug will be added to the list to fix

3djc, 2016-09-19:
[image]

I do not seem to get you issue

Bumblebee78, 2016-09-19:
[image]

[image]

so it looks at the Horus

3djc, 2016-09-19:
What nightly are you running ?

3djc, 2016-09-19:
You are not using the timer widget, that's why

Bumblebee78, 2016-09-19:
OpenTX 2.2.0 RC5 (N357)

3djc, 2016-09-19:
[image]

Bumblebee78, 2016-09-19:
Oh..with the Timer widget works.

kilrah, 2016-09-19:
Then.. where did it not work? How should we guess?

3djc, 2016-09-19:
And the timer limits to 1 hour should be extended in N358

Closing this if you don't mind

closed 2016-09-19 by 3djc