opentx/opentx #3832: Tele option is  false when no telemetry is available

fraternl, 2016-09-19:
To check if an Frsky receiver is listening I can now use the "Tele" boolean in logical switches.
I don't even have to put it in a model as I can use it in a Global Function as well.

I let my Horus beep every 5 seconds to alert me that there's no receiver listening.

Today I received my Banggood multi-protocol module and I used it to fly my Floureon H101 nano-drone after selecting the Bayang-protocol.
Too bad it's now also giving me that beep every 5 seconds...

I know that some special functions don't work under some conditions.
Can this Special Function with "Tele" stop working if telemetry is not possible???

kilrah, 2016-09-20:
That would be a nightmare since that changes every week with the multimodule.

You can disable global functions for a single model.

fraternl, 2016-09-20:
The global function disable is a workaround, yes... 
It is very hard to use those global functions though...
Maybe when we get global switches....

closed 2016-11-27 by fraternl