opentx/opentx #3836: Problem reading last sectors on the SD card

projectkk2glider, 2016-09-20:
Use commit 444d552d5a9ecaba1739609cd8af41e0dbea7f24 to test/reproduce:

* radio Horus
* compile with CLI and DEBUG enabled
* execute CLI command **testsd**

Results, 8GB SDHC card:
```
>testsd
SD card has 15554560 sectors
Starting single sector read test, reading 16 sectors one by one
sector 15554544 read OK
sector 15554545 read OK
sector 15554546 read OK
sector 15554547 read OK
sector 15554548 read OK
sector 15554549 read OK
sector 15554550 read OK
sector 15554551 read OK
sector 15554552 read OK
sector 15554553 read OK
sector 15554554 read OK
sector 15554555 read OK
sector 15554556 read OK
sector 15554557 read OK
sector 15554558 read OK
sector 15554559 read OK

Starting multiple sector read test, reading two sectors at the time
Status(WaitRead)=28, s:15554544 c: 2
sector 15554544-15554545 read OK
Status(WaitRead)=28, s:15554546 c: 2
sector 15554546-15554547 read OK
Status(WaitRead)=28, s:15554548 c: 2
sector 15554548-15554549 read OK
Status(WaitRead)=28, s:15554550 c: 2
sector 15554550-15554551 read OK
Status(WaitRead)=28, s:15554552 c: 2
sector 15554552-15554553 read OK
Status(WaitRead)=28, s:15554554 c: 2
sector 15554554-15554555 read OK
Status(WaitRead)=28, s:15554556 c: 2
Status(WaitRead)=28, s:15554556 c: 2
Status(WaitRead)=28, s:15554556 c: 2
sector 15554556-15554557 read FAILED, err: 1
Status(WaitRead)=28, s:15554558 c: 2
Status(WaitRead)=28, s:15554558 c: 2
Status(WaitRead)=28, s:15554558 c: 2
sector 15554558-15554559 read FAILED, err: 1

Starting multiple sector read test, reading 16 sectors at the time
Status(WaitRead)=28, s:15554544 c: 16
Status(WaitRead)=28, s:15554544 c: 16
Status(WaitRead)=28, s:15554544 c: 16
sector 15554544-15554559 read FAILED, err: 1
```

Results, 2GB SD card (non-HC):
```
SD card has 3887104 sectors
Starting single sector read test, reading 16 sectors one by one
sector 3887088 read OK
sector 3887089 read OK
sector 3887090 read OK
sector 3887091 read OK
sector 3887092 read OK
sector 3887093 read OK
sector 3887094 read OK
sector 3887095 read OK
sector 3887096 read OK
sector 3887097 read OK
sector 3887098 read OK
sector 3887099 read OK
sector 3887100 read OK
sector 3887101 read OK
sector 3887102 read OK
sector 3887103 read OK

Starting multiple sector read test, reading two sectors at the time
Status(WaitRead)=28, s:3887088 c: 2
sector 3887088-3887089 read OK
Status(WaitRead)=28, s:3887090 c: 2
sector 3887090-3887091 read OK
Status(WaitRead)=28, s:3887092 c: 2
sector 3887092-3887093 read OK
Status(WaitRead)=28, s:3887094 c: 2
sector 3887094-3887095 read OK
Status(WaitRead)=28, s:3887096 c: 2
sector 3887096-3887097 read OK
Status(WaitRead)=28, s:3887098 c: 2
Status(WaitRead)=28, s:3887098 c: 2
Status(WaitRead)=28, s:3887098 c: 2
sector 3887098-3887099 read FAILED, err: 1
Status(WaitRead)=28, s:3887100 c: 2
Status(WaitRead)=28, s:3887100 c: 2
Status(WaitRead)=28, s:3887100 c: 2
sector 3887100-3887101 read FAILED, err: 1
Status(WaitRead)=28, s:3887102 c: 2
Status(WaitRead)=28, s:3887102 c: 2
Status(WaitRead)=28, s:3887102 c: 2
sector 3887102-3887103 read FAILED, err: 1

Starting multiple sector read test, reading 16 sectors at the time
Status(WaitRead)=28, s:3887088 c: 16
Status(WaitRead)=28, s:3887088 c: 16
Status(WaitRead)=28, s:3887088 c: 16
sector 3887088-3887103 read FAILED, err: 1
```

Test results are repeatable, multiple executions of the **testsd** produce the same result.

I want to confirm this issue on at least one another radio. Mine is pre-beta unit, but should be pretty close to beta units.

3djc, 2016-09-20:
SD card has 7821312 sectors
Starting single sector read test, reading 16 sectors one by one
sector 7821296 read OK
sector 7821297 read OK
sector 7821298 read OK
sector 7821299 read OK
sector 7821300 read OK
sector 7821301 read OK
sector 7821302 read OK
sector 7821303 read OK
sector 7821304 read OK
sector 7821305 read OK
sector 7821306 read OK
sector 7821307 read OK
sector 7821308 read OK
sector 7821309 read OK
sector 7821310 read OK
sector 7821311 read OK

Starting multiple sector read test, reading two sectors at the time
Status(WaitRead)=28, s:7821296 c: 2
Status(WaitRead)=28, s:7821296 c: 2
Status(WaitRead)=28, s:7821296 c: 2
sector 7821296-7821297 read FAILED, err: 1
Status(WaitRead)=28, s:7821298 c: 2
Status(WaitRead)=28, s:7821298 c: 2
Status(WaitRead)=28, s:7821298 c: 2
sector 7821298-7821299 read FAILED, err: 1
Status(WaitRead)=28, s:7821300 c: 2
Status(WaitRead)=28, s:7821300 c: 2
Status(WaitRead)=28, s:7821300 c: 2
sector 7821300-7821301 read FAILED, err: 1
Status(WaitRead)=28, s:7821302 c: 2
Status(WaitRead)=28, s:7821302 c: 2
Status(WaitRead)=28, s:7821302 c: 2
sector 7821302-7821303 read FAILED, err: 1
Status(WaitRead)=28, s:7821304 c: 2
Status(WaitRead)=28, s:7821304 c: 2
Status(WaitRead)=28, s:7821304 c: 2
sector 7821304-7821305 read FAILED, err: 1
Status(WaitRead)=28, s:7821306 c: 2
Status(WaitRead)=28, s:7821306 c: 2
Status(WaitRead)=28, s:7821306 c: 2
sector 7821306-7821307 read FAILED, err: 1
Status(WaitRead)=28, s:7821308 c: 2
Status(WaitRead)=28, s:7821308 c: 2
Status(WaitRead)=28, s:7821308 c: 2
sector 7821308-7821309 read FAILED, err: 1
Status(WaitRead)=28, s:7821310 c: 2
Status(WaitRead)=28, s:7821310 c: 2
Status(WaitRead)=28, s:7821310 c: 2
sector 7821310-7821311 read FAILED, err: 1

Starting multiple sector read test, reading 16 sectors at the time
Status(WaitRead)=28, s:7821296 c: 16
Status(WaitRead)=28, s:7821296 c: 16
Status(WaitRead)=28, s:7821296 c: 16
sector 7821296-7821311 read FAILED, err: 1

projectkk2glider, 2016-09-20:
@3djc Hehe, yours in even worse than mine. I have a work-around already in the works.

projectkk2glider, 2016-09-20:
With f03837f45d61733b1d363d82c7455d33a3b34e8e success:
```
SD card has 3887104 sectors
Starting single sector read test, reading 16 sectors one by one
sector 3887088 read OK
sector 3887089 read OK
sector 3887090 read OK
sector 3887091 read OK
sector 3887092 read OK
sector 3887093 read OK
sector 3887094 read OK
sector 3887095 read OK
sector 3887096 read OK
sector 3887097 read OK
sector 3887098 read OK
sector 3887099 read OK
sector 3887100 read OK
sector 3887101 read OK
sector 3887102 read OK
sector 3887103 read OK

Starting multiple sector read test, reading two sectors at the time
sector 3887088-3887089 read OK
sector 3887090-3887091 read OK
sector 3887092-3887093 read OK
sector 3887094-3887095 read OK
sector 3887096-3887097 read OK
Status(WaitRead)=28, s:3887098 c: 2
Status(WaitRead)=28, s:3887098 c: 2
Status(WaitRead)=28, s:3887098 c: 2
disk_read() multi-block failed, trying single block reads...
sector 3887098-3887099 read OK
Status(WaitRead)=28, s:3887100 c: 2
Status(WaitRead)=28, s:3887100 c: 2
Status(WaitRead)=28, s:3887100 c: 2
disk_read() multi-block failed, trying single block reads...
sector 3887100-3887101 read OK
Status(WaitRead)=28, s:3887102 c: 2
Status(WaitRead)=28, s:3887102 c: 2
Status(WaitRead)=28, s:3887102 c: 2
disk_read() multi-block failed, trying single block reads...
sector 3887102-3887103 read OK

Starting multiple sector read test, reading 16 sectors at the time
Status(WaitRead)=28, s:3887088 c: 16
Status(WaitRead)=28, s:3887088 c: 16
Status(WaitRead)=28, s:3887088 c: 16
disk_read() multi-block failed, trying single block reads...
sector 3887088-3887103 read OK

```

closed 2016-09-20 by projectkk2glider

projectkk2glider, 2016-09-20:
I also verified the new read routine. I wrote distinct patterns into the last two SD card sectors (with radio in mass-storage mode connected to computer) and then read back those two sectors in CLI. Both when reading single sector and when doing multi-block read (single block fall-back was used) the contents were read correctly.

commit f03837f4 (2016-09-20): Fixes #3836: if multi-block SD read fails, fall back to the single block reads [Horus]