opentx/opentx #3837: LUA Wizard no longer works in 2.2.0N358 (Taranis+)

lichtl, 2016-09-20:
Taranis + - opentx-x9d+-lua-mixersmon-multimodule-ppmus-en-2.2.0N358

After moving wizard folder into scripts (SD card folder structure seem to be wrong / sdcard.h might need adjustment) the wizard just return an error when create a new model.

```
Script syntax error
wizard.lua:72
attempt to call field 'lock' (a nil value)
```

I also noticed that between 2.1 and 2.2 some SD-card folder names have changes (BMP > IMAGES), (FIRMWARES > FIRMWARE), (EEPROMS > EEPROM). Maybe it would also make sense to create those empty folder in the SD-card structure (including the SCRIPTS subfolder) to reduce confusions.

kilrah, 2016-09-21:
The script has not been ported yet indeed.

closed 2016-10-16 by lichtl

hketterer, 2016-11-21:
I have 2.20n359 installed on my taranis plus and still get:
Script syntax error
wizard.lua:72
attempt to call field 'lock' (a nil value)

kilrah, 2016-11-21:
Are you sure you have the wizard for 2.1 loaded, not the one for 2.0?

hketterer, 2016-11-21:
i loaded the files from here http://downloads.open-tx.org/2.2/nightly/sdcard/

hketterer, 2016-11-21:
you got it for me....i deleted the scripts folder and copied the folder from the zipped file onto my sd card and it works. dont know what the problem was cuz the original firmware was 2.1.8. so it already would have had the 2.1 wizard. and i copied the whole sd card from the link when i got an sd card version error.
but all is good.
thank you all for the hard work you do for us. without all of you most of us loaly users wouldnt make it through all the upgrades without pulling all our hair out.

commit adb4b122 (2016-10-16): Fix #3837 - Fix wizard for non Horus (#3902)

* move WIZARD in correct folder

* remove lcd.lock function