opentx/opentx #3839: [Enhancement][OpenTX 2.1.x] Force PPM input on trainer/DSC port

LapinFou, 2016-09-21:
Hi folks,

I tried to do this modification on my Taranis in order to get an integrated wireless buddy box: http://flitetest.com/articles/wireless-buddy-box

I've done this wiring:
[image]

Unfortunately is not working unless there is a jack plugged in the DSC port.
According this thread on RCG, it was working on OTX 2.0: http://www.rcgroups.com/forums/showthread.php?t=2149001

I found this pic (Graupner MX16):
[image]

The DSC plug looks very similar. So, I guess the control on the "SW" side was not taking in account on OTX2.0, but is on OTX2.1?

Here is the enhancement request: Add a special function to force OTX to use the PPM input of the DSC port even if not jack plug is detected.
Is it possible?

kilrah, 2016-09-21:
Correct, the signal is ignored if no plug is in as that noise on the trainer line would lead to false trainer activations.
It's not going away, given you're into soldering in the radio anyway you can just force the contact to be always active if you want.

closed 2016-09-21 by kilrah

LapinFou, 2016-09-21:
Hi Kilrah, I think I wasn't clear enough.
I never asked to remove this protection. This will definitely be a bad idea.
I was requesting to have a dedicated "special function" to bypass this protection for people doing the DIY mentioned below. 
In the same way, you got the DSM mode for the hack module or the LP45 modulation.
I believe this enhancement proposal is in the OpenTX's philosophy, don't you think?

kilrah, 2016-09-21:
Again I don't think this warrants the hassle of a confusing software solution since the hardware one is easily done while you're there to do the mod in the first place (just cut a trace, I believe).

LapinFou, 2016-09-22:
OK. I don't really agree, but I understand and respect your choice. :smile: 
I will try to cut a track and see how it does work.
By any chance, do you have the schematic of this back board?

bobtitus, 2017-07-30:
I agree with LapinFou, it would be a good thing to have a function to deactivate the protection at will without having to fiddle with a jack. It was just fine on 2.0, to me it's more a downgrade than anything else.