opentx/opentx #3840: USB connection terminates after approx. 15 s

Hartinger, 2016-09-21:
With OpenTx 2.1.9 on my Taranis Plus (mass storage option enabled), the SD card appears as drive F: when I connect the radio to the USB during normal operastion, as expected.
However, after approx. 15 s, the drive disappears without any interaction, and a click can be heard (similar to the one you hear on exiting the Bootloader).
When I changed back to version 2.1.8, the problem disappeared.
Re-flashing 2.1.9 made the problem reappear.

projectkk2glider, 2016-09-21:
Could you try the same connection from the bootloader.

Hartinger, 2016-09-21:
From the bootloader (latest version 2.1.9) the USB connection works fine, both drives (F: and G: in my case) appear and remain stable.

lshems, 2016-09-30:
Did you try copying some files to the card. I have the same problem, but when writing to the card, it will stop in bootloader mode as well.

kilrah, 2016-09-30:
It's certainly the watchdog, we fixed that on Horus a few weeks ago but didn't check the Taranis.

ndoty, 2016-10-02:
Not sure if this is the same exact bug, but when Mass Storage is enabled in the firmware the radio when already on, will freeze when a USB cable is plugged in. I use my Taranis as a PC Joystick and i had to re-flash with out the mass storage option checked in order for it to work.

3djc, 2016-10-02:
Yours is not an issue ndoty, you cannot have joystick and mass storage at the same time.

projectkk2glider, 2016-10-02:
Just for reference OpenTX 2.2 on Taranis (using my branch 40aa791a03299875d746aa6c3360f2bf6fafd145) works both in MASSSTORAGE and Bootloader without any problems (using Linux). 

I will now try the 2.1.9.

projectkk2glider, 2016-10-02:
Confirmed on 2.1.9, the radio gets reset after a while and enters ST Bootloader (DFU). 

A fun fact: after the incident, even after multiple power on/off cycles the bottom part of the LCD (about 4 lines) was corrupted. To fix it I had to remove the battery.

closed 2016-10-02 by projectkk2glider

lshems, 2016-10-03:
Just to inform: I have this issue on a taranis plus, no matter what firmware or bootloader. So probably hardware related topics n my case.

If I ever solve it, I'll post the findings.

commit f38963d6 (2016-10-02): Fixes #3840: Radio was reset 20s after USB connection in MASSSTORAGE mode (broken in 7fd412f7195, fix ported from a33b1d2659)