opentx/opentx #3841: Companion 2.2N358 Simulator crash

lichtl, 2016-09-21:
The Simulator for Taranis+ hangs straight after start. I've tried several installation under Windows 7 + Windows 10.

Uploading a Video at the moment.

kilrah, 2016-09-21:
No need for a video, we're aware of it.

projectkk2glider, 2016-10-02:
Can't reproduce with 29e618146c673f605234a19210157c244c7d9cd4 on Linux.

lichtl, 2016-10-11:
Just tested 7e8bebfaaccaa89f86890717df681a25d4f164e3 under Linux (Ubuntu) and noticed the same behaviour. Simulator got stuck at the start-screen

lichtl, 2016-10-11:
Just some more testing. It seems the simulator only works if started directly. Start the simulator via companion22 will cause it to hang/crash.

Heliobteixeira, 2016-10-13:
Even starting simulator alone wont solve the issue for me on Windows 7 crashing after wrong eeprom data message.

projectkk2glider, 2016-10-13:
Ok, reproduced on Linux with *Release* build. And it is not a crash, but unresponsive and stuck simulator window.

projectkk2glider, 2016-10-13:
Ha ha, nice bug, this is how eeprom thread is started:
```c
  assert(!pthread_create(&eeprom_thread_pid, NULL, &eeprom_thread_function, NULL));
```
In release build this line does nothing. :roll_eyes:

projectkk2glider, 2016-10-13:
@lichtl can you test and confirm my fix?

lichtl, 2016-10-13:
@projectkk2glider looking good at a fresh compiled linux companion. Is there somewhere a windows binary available to test as well?

bsongis, 2016-10-13:
Funny bug, could be mine :)

bsongis, 2016-10-13:
I suppose you git pruned?

fred99, 2016-10-13:
I have recompiled with your fix under W10/MVS14 with the Release build and now the simulator works like a charm in the 2 modes, direct and from Companion : no more hang

lichtl, 2016-10-14:
Sweet. 👍

commit 0c26cd07 (2016-10-13): Fixes #3841: simulator EEPROM thread was not created in Release build
closed 2016-10-14 by lichtl