opentx/opentx #3845: Timers with ThS and ThS should look on throttle input, not throttle stick

hsteinhaus, 2016-09-22:
Currently the timers are controlled by the throttle stick, but not by the throttle input as defined in menu "inputs". This seems to be illogical, especially when moving the throttle control to other switches/sliders/knobs.

Use case: consider a powered glider, equipped with both motor and air brakes. Usually the brakes are placed on the throttle stick, while the motor is put on a switch. You don't need anything between on and off in a glider. Sometimes, intermediate throttle settings are even forbidden and would instantly kill the ESC, e.g. with ultra-high power drives used in F5B.

kilrah, 2016-09-22:
It looks at the configurable "throttle source", and common practice is to set the output channel used for throttle.

closed 2016-09-22 by kilrah

hsteinhaus, 2016-09-22:
Cannot confirm this. At least with 2.1.8 and a throttle source set to "SF" the timer is definetely following the stick, not the switch configured as trottle source.