opentx/opentx #3847: Enable the "bind" special function in nightlies or make it optional

fraternl, 2016-09-22:
Now that we have multiprotocol support in 2.2 it becomes tedious that I need to bind my nano-drone everytime I swap the batteries.
I've enabled the autobind option which binds the 1st flight, but on all consecutive flights I have to go through the model menu and bind it.

I would rather use a logical switch that can also take care of my timers at the same time.
This way I can keep track of the total time flown that evening and the current flight.

I know the code is already made, but I can only enable it if I compile openTX myself.
A build option for this in Companion has not been made.

schwabe, 2016-09-23:
(fraternl is talking about the DANGEROUS_MODULE_FUNCTIONS)

fraternl, 2016-09-25:
...And fraternl doesn't think it's that dangerous at all....   ;-)

fraternl, 2016-11-10:
I would like to add to this that multi-protocol module owners are now running a backported 2.1 release where this option is enabled.
They will be in for a surprise when their bind function suddenly isn't working anymore after they upgrade to 2.2

Dutchy-79, 2016-11-10:
Yes, this is exactly what I need. 
I want my momentary switch to reset timer 1 and initiate the binding process when swapping battery from my toy quad

nigelsheffield, 2016-11-10:
Plus one, i am using this exact function on my quads built with h101 flight controllers.
They need auto bind every time I replace the battery.
Note that this would be an option tick box which is off by default so only those that need this would tick the box before downloading .
More and more of us are using toy quad control board's to fly small home built quads, it makes the build much smaller and cheaper and often more flexible then using standard flight controller.

spikey1973, 2016-11-11:
plus another one. 

i completely adhere to nigelsheffields comment, specially about the tick box (maybe even make an extra allert when ticking the box about the possible dangeres) but one should have at least an option to do so.

TinuZ-s, 2016-11-14:
With all the protocols needing a re-bind as soon as you swap battery it would be better for my On/Off switch's life to bring have DANGEROUS_MODULE_FUNCTIONS as an option without having to build ourselves. We are grown up boy's and girls and I think we can decide to use it or not. Nigel's idea makes totally sense. Let us choose to have it in our firmware or not.

We can do already potentially dangerous programming which would cause immediate loss of control in so many ways. This is just another one.

projectkk2glider, 2016-11-14:
I would prefer a solution where we add a new firmware option in Companion and corresponding handling in build scripts. Enabling this sistem-wide could bite innocent users.

3djc, 2016-11-14:
And there may be a need for be a warning at start I think, for the innocent friend who 'know what's best for them' and install them firmware with dangerous settings without telling them

schwabe, 2016-11-25:
Or we could add a check in the general model settings that says "Enable dangerous model settings" that will display a warning message if you activate it. (EEPROM changes issues aside)

OpenPanzerProject, 2016-11-25:
Another plus one.

fraternl, 2016-11-26:
I was made aware of a workaround for the "binding problem" 
A "reset flight" in combination with "autobind" has supposedly the same effect.
I didn't test this.
The module is apparently turned off momentarily.
That's contrary to a previous rocket chat discussion where it was told that reset flight didn't turn off the module. It was then more about the check screen afterward that would have the module turned off and could invoke a failsafe condition that was programmed by the radio into the transmitter. 

Doing some "reverse reasoning" here... Doesn't that make "reset flight" a "dangerous module function"?

schwabe, 2016-11-26:
@fraternl on Taranis reset flight it does that. On that horus it doesn't

Dutchy-79, 2016-11-26:
I withdraw from the discussion, the developers are aware of the question and eventually it will somehow be solved. 

Like I said on rcgroups it should be possible to do this safe. 

Make a few logical and physical switches that do the following. 

Switch A 100% is blocked throttle and bind possible. 

Switch A - 100% is blocked bind switch and throttle unlocked. 

If you wanna go super safe you can add a switch B so A and B must be 100% or if A = 100 and B = - 00,....

TinuZ-s, 2016-11-26:
I still do not see why anyone would assign a function if they would not know what the function does. The discussion about that fact is moot. 
Logic dictates that you would only choose what you want/need, and not randomly 'something' from a selection list. Analog to this: Why would I want to reduce my rates to 0 on all primary channels in low rates.. pretty dangerous function as well, yet perfectly acceptible to the devs. 

The devs are not here to make policy against idiocy.

closed 2016-11-27 by fraternl

nigelsheffield, 2017-10-05:
Has this ever been implemented, I still cannot build my own version of opentx and am stuck on a old version of 2.1.9 which someone built for me with the binding function....
I really want to upgrade to 2.2 and stay current ,
I know I am not alone in this.
Please can we have it as a tick box?

schwabe, 2017-10-05:
@nigelsheffield no, there won't be a tick, also if you want that for multi, there is no need as multi now has a bind on channel 16 functionality that does the same.

nigelsheffield, 2017-10-05:
Great thanks, so I just set Ch 16 in the tx to bind , yes I do remember reading something about that now.
Sorry to bother you,
 thanks!