opentx/opentx #3852: Bluetooth not advertising on Horus

stytom, 2016-09-24:
On Horus, BLE Module is enabled and named but does not advertise any data and is thus not visible.
In addition when trying to set serial port to S-port mirror the radio falls into Emergency Mode

kilrah, 2016-09-24:
Serial hasn't been taken care of yet indeed.

BT is unused on both X9E and Horus at this point. No time could be found for that so far.

dk7xe, 2017-03-08:
Would be nice to get BT working since 
- a cable might not be available always.
- there are still issues with files going corrupt when USB cable is pulled out while Horus is still on.

1Sven, 2017-04-30:
Yes it will be nice. Filetranfer and Trainer...

xtrmtrk, 2017-07-19:
And sending out telemetry data over BT to Tower or similar ground station apps.

MarenB, 2018-08-26:
Hi!

I am pretty sure, BT was working on my X9E in version OpenTX 2.0.x 
After I updated to 2.2.1 I have two issues:
1. BT cannot be enabled from the radio, only via Companion
2. if I enable it, there's no telemtry data send on BT anymore

Unfortunately, my antenna tracker is based on the BT data stream. The BLE module of the tracker connects, but there's no usable data received,

MarenB, 2018-08-30:
Hi Guys,

I can now confirm: something happened to the BT datastream between OTX 2.1.9 and 2.2.x

My BT master receives the smartport telemetry without any issues from my X9E with 2.1.9, after updating to 2.2.x it seems to receive empty or no data.

closed 2018-08-30 by kilrah