opentx/opentx #3853: Wrong value read out

defnotben, 2016-09-25:
hey guys i had assigned a switch to read out the telemetry value of the voltage on board the drone, but everytime i toggle the switch, it reads the voltage with an additional o.5 volts. for e.g. if its 11.4v, it'll read 11.9v.

but when its above 11.5, e.g. 11.7, after it adds 0.5 itself, it'll round up to 12 volts. Any advice? :")

3djc, 2016-09-25:
Where does the value come from ? Frsky sensor ?

kilrah, 2016-09-25:
The voice files on your SD card to not match your OpenTX version. Load the correct ones.

closed 2016-09-25 by kilrah

defnotben, 2016-09-25:
It comes from the flight controller, connected to xsr receiver via S port... The value shows correctly in the transmitter, but when I assign a switch to read out the value, the +0.5 in value happens :(

defnotben, 2016-09-25:
I don't have any sd card inside the taranis.. Any advice?

kilrah, 2016-09-25:
If you didn't you'd have no voice at all.

defnotben, 2016-09-25:
ohhh i understand what u mean now haha, but how do i change the voice files?