opentx/opentx #3855: 9xr with gvars - cannot set negative value in Special Functions

KushnirenkoTaras, 2016-09-25:
**Tansmitter** Turnigy 9XR
**Firmvare** opentx-9xr-audio-gvars-en-2.1.9.hex

If i add in Special Functions line like 

`THR Adjust GV1  nnn   v`

**nnn** can be set from 0 to 125

kilrah, 2016-09-27:
That is normal, there are only 8 bits available to store a value in this function.

KushnirenkoTaras, 2016-09-27:
This not normal.
8-bit value can be considered as a signed number.
Thus, we have a range of values from -128 to 127

schwabe, 2016-09-27:
Some of the values are reserved for internal purpouses.

KushnirenkoTaras, 2016-09-27:
Then you need to fix OpenTX Companion.
In the Companion is possible set the values in the range -125 125
But it is useful to be able to set negative values in 9XR.
Similarly, the Companion.
Otherwise, global variables to me useless.

KushnirenkoTaras, 2016-09-28:
Strange approach.
Instead to improve the functionality simply decided to cut.
It's a pity.

kilrah, 2016-09-28:
The limitation stems from the very limited resources on the 9x hardware, improving that may be possible but would likely represent more work than what current developers are willing to invest in a dying platform, the 9X platform in OpenTX has basically been untouched for 2 years already. 

Feel free to find an acceptable solution and implement it, that's what open source is for. 

Note that you have ways around the limitation, i.e. using a source (like an unused channel) to set the GVAR value.

closed 2019-08-19 by kilrah