opentx/opentx #3858: SF "increment/decrement GV" doesn't work as intended in 2.2n

strgaltdel, 2016-09-28:
Hi,
just made some migration tests to 2.2n358

oTx 2.18 was able to use GVs as a counter by define some SFs like this

[image]

so the stepping rate was +/-1 

after converting to 2.2, the GV value jumps quickly to its new defined max/min values (+/- 1024 as standard?) by pulling/pushing the configured switch.
incremental value seems to be completely ignored

regards
Udo

Viruswelt, 2016-09-30:
i have the same problem with the Trim Switches. i use these switches on taranis to ingrease/degrease a Gv +1/-1 to ajust a gyro gain.

kilrah, 2016-10-16:
Set/Reset SFs have the same problem

fraternl, 2016-10-17:
The action itself apparently requires an interval to prevent a race condition, but the SF itself could be expanded in a way that it works the same as "Play Track" where an interval can be given.  !1x 1x 1s 2s 3s instead of the enable/disable checkbox.

This also means we would lose the enable/disable option.
This could be solved by adding a 0x (zero times).

projectkk2glider, 2016-10-17:
The SF reset problem is the same in 2.1 release! I wonder how that went unnoticed. :worried:

3djc, 2016-10-17:
I think it could make sense to apply the same options as play track to a few more fonctions. For some, it is unclear how they should behave when coupled with a non edge  trigger (screenshot,...)

fraternl, 2016-10-17:
If I would have simplified my setup a year ago, I would have noticed then.
It did take me several rows of LS's

L1  LowC < 3.7 Delay(5s)
L2 LowC > 4.18 Delay(5s)
L3 Sticky(L2, L1)
L4 Edge(L3, [1:instant]) 

SF1 L4 - Reset(Flight)
SF2 L4 - Reset(Cnsp)
SF3 L4 - Play Track(reset) 1x

projectkk2glider, 2016-10-17:
Most of the functions that need it have an activation flag that handles them correctly - ie only fire once when the switch becomes true. One such function is Screenshot. Some are missing this processing step.

3djc, 2016-10-17:
Yes, but what you might want to do by using a 'constant' on trigger, il to take more than one. Rather than decide which one should be runed once or not, I thjink it could make sense to leave that choice to the user in the same way as playtrack does

kilrah, 2016-10-17:
If you want repeated events you can use a Timer LS, and have even more precise control.

strgaltdel, 2016-10-23:
Just to get confirmed:
the solution has not made his way into RC7, right?
I have to wait for RC8 or make an interims compiling?

Today i got the same wrong behavior after update to nightly 359 (Taranis X9e)

kilrah, 2016-10-23:
Yes it has and should be fixed.

strgaltdel, 2016-10-23:
simple test (Rc7):
[image]

under 2.18 this config will increment GV9 every time switch SH is pushed by value "1"

otx 2.2 RC7:
this config will increment GV9 every time switch SG is pushed, directly from "0" to value "1010" !
The options "no repeat" etc doesn't affect the result

kilrah, 2016-10-23:
Try it on the radio itself, not in the simulator

tomkfly, 2016-10-23:
When I tried it on the radio the GVAR increment end by 10 instead of 1.

projectkk2glider, 2016-10-23:
I will re-test this now.

projectkk2glider, 2016-10-23:
Taranis `simu` is OK.
Taranis Compaion radio simulator is OK (btw Companion model editor is actually broken, the enable option is missing for the GVAR INC/DEC SF)
Horus radio is OK.

I can't reproduce.

strgaltdel, 2016-10-23:
i made my tests on an existing model with following "history":
oTx 2.18 >>RC6 >>RC7
here the GV functionality seems totally "messed up", every GV shows deifferent, but "fixed" values....
BUT
When i build a new model the same SF-line settings are working well

projectkk2glider, 2016-10-23:
Wait, I have something....

projectkk2glider, 2016-10-23:
Companion is the only broken thing here, it seems it injects a repeat parameter somehow. The thing actually works OK on the radio. On my system the GVAR is incremented every second while the switch is true.

@strgaltdel plese repeat the test with a new empty model programmed on the radio.

strgaltdel, 2016-10-23:
yepp,
as i wrote before, by using an empty model all is working fine
i will do some more researches why the "transformed" models are not working as expected, 

thanks !

projectkk2glider, 2016-10-23:
Damn, this is strange, can't even reproduce it now in Companion. 

Looks like the bug is originating from e8aaa674505946a5b1009b47c7ddccdf9c5457a7 were the GVAR INC has *repeat and enable* parameters in Companion, while this combination is not implemented on the radio. Additionally the GVAR INC can't have a repeat parameter because of `HAS_REPEAT_PARAM(func)` and `HAS_ENABLE_PARAM(func)` macros.

So to fix this we have two options:
 * drop the broken repeat support for GVARS SF (my preferred solution)
 * or implement it fully

@bsongis and @kilrah please advise.

kilrah, 2016-10-23:
Agree on dropping repeat, that was never intended in the first place and most likely a bug.

projectkk2glider, 2016-10-23:
Actually this merge commit cbec010f44adf078ef8c7694d9d398737d670dc3 wiped out the HAS_REPEAT_PARAM(func) and HAS_ENABLE_PARAM(func) macros.

projectkk2glider, 2016-10-23:
So it looks like @bsongis moved GVAR SF to repeat. This is also strongly related to #3770.

projectkk2glider, 2016-10-26:
I believe that @bsongis idea was good, so I will try to re-implement it (hopefully without the bugs).

commit 85f7a09b (2016-10-18): Fixes #3858: added edge detection to RESET and GVAR INC/DEC functions (#3921)

* Fixes #3858: added edge detection to RESET and GVAR INC/DEC functions

* Re #3858: Gtests added for Special Functions: FLIGHT_RESET and GVAR_INC

* Re #3858: Gtests fixed

* Re #3858: Gtests fixed 2
commit e9a06090 (2016-10-18): Fixes #3858: added edge detection to RESET and GVAR INC/DEC functions (ported from next) (#3920)
commit 8626c072 (2016-10-27): [Companion] Fixes #3858 (#3958)
commit 45dc76bf (2017-06-25): Avoid introducing #5036 in any future 2.1 release

Revert "Fixes #3858: added edge detection to RESET and GVAR INC/DEC functions (ported from next) (#3920)"

This reverts commit e9a06090dd5f040b68d9adfb9f71434dd05785cb.
closed 2016-10-27 by projectkk2glider