opentx/opentx #3860: Open TX Horus 2.2 ON352 - Extendended trims cause  display and system to freeze

clivemartin, 2016-09-29:
When extended trims selected and when trim (t5 as example) is held in upward direction until limit is reached the trim curser moves into top bar of display, at this point the display and OS freezes.  Only way to recover situation is to hold on/off switch for 10-15 seconds to switch system off and reset.

kilrah, 2016-09-29:
Is that only for T5/T6 or also the normal ones?

clivemartin, 2016-09-29:
Just been trying replicate fault of which I can do , I’m sorry but I should have reported T2 not T5.

However to give more detail, first of all I use cross trims, and the redundant T2 was set up to adjust the volume of the system and I have extended trims set. Whilst setting the maximum volume using T2 the trim curser travelled into the upper most screen and both screen and OS froze.  I rebooted the system, unchecked the extended trim option and reported issue.

I have now rechecked the extended trim option to replicate fault and to give more clarity, by the moving trim T2 to max (forward) releasing trim when max achieved and then setting trim further upwards to extended trim position, the curser moves into upper screen and when maximum travel is reached the system freezes. No other functions available. On reboot all trims return to centre position. Same fault appears on T3, Even stranger things happen when T4 and T1 are activated !!! the trim curser for these trims travel across the centre of the screen (flashing) but system does not freeze.

kilrah, 2016-09-29:
Thanks for the additional info

closed 2016-10-15 by kilrah