opentx/opentx #3867: D Series Sensors not showing up on Horus N258

flyingdreamz, 2016-10-01:
The voltage sensor worked in 2.1 but in 2.2 on the horus, the horus is not finding them. This is the case on my D8R-XP with the FBVS-01 and FAS-40. As well as my X6R using the FBVS-01.

kilrah, 2016-10-02:
You need to "discover" them. Definitely works, I'm using some.

closed 2016-10-02 by kilrah

flyingdreamz, 2016-10-03:
I have been unable to get the discover function to find them. So far I am at 8 Sensors that the Horus is not seeing.

3djc, 2016-10-03:
It clearly works, there must be something wrong on your side:
[image]
[image]

projectkk2glider, 2016-10-03:
I think OP has a problem with sensors that connect trough the telemetry connector (via HUB). I would not be surprised if IXJT does not handle external HUB sensors somehow. 

@flyingdreamz do you get at least RSSI, A1 and A2 sensors discovered and working?

flyingdreamz, 2016-10-04:
I get RSSI but I do not get A1 or A2.
I have literally changed nothing between my Opentx 2.1 on the T+ to my Horus. In fact on the X6R using the Analog port with https://alofthobbies.com/frsky-battery-voltage-sensor-fbvs-01.html it has been direct soldered for a year. When I rebind with my T+ the sensor shows up, when I bind to the Horus I just get RSSI and RX Voltage. On my D8R-XP I get no telemetry other then RSSI.

3djc, 2016-10-04:
Wait, I'm unsure of what you are expecting, but an X6R will report RSSI and Rxbt. Extract of X6R docs:
"When pins of A1 and X (Middle Pin) are connected by the jumper, A1 will change from External Analog
Telemetry Port to Internal Built-in battery voltage sensor."

Here is a X6R with FVBS:
[image]
[image]

Remove it and place the jumper:
[image]
[image]

I'm unsure of what else you are expecting

projectkk2glider, 2016-10-05:
Example of D series receiver with a lot of sensors - working:
[image]