opentx/opentx #3868: Sound System crash - Information

pau33, 2016-10-02:
Only an information, I cannot reproduce this error so far. During a flight this weekend, I had a total crash of the sound system (Taranis Plus). No more sounds not on the internal speaker and not on the headphone. Fortunately the rest works perfect including the telemetry on the radio. I landed, switched the radio off and on again and everything worked again. I did no flash the last 2 month, this was the first occurrence.

projectkk2glider, 2016-10-02:
Which OpenTX version? 

Most likely a duplicate of #1838. The work in #3815 should at least improve this.

pau33, 2016-10-02:
It is the actual 2.1.9 Version. It happens during flight, not at startup. So it shouldn't be the SD-Card? As I can see, there will be improvements in the 2.2.x. Thank you very much. If the problem occurs again, I will report.

3djc, 2016-10-03:
I have had this previously with myX9E, it was the SDCARD (exact same symptoms as yours)

closed 2016-10-03 by projectkk2glider