opentx/opentx #3869: Horus Completely Sound System 3.10.2016

Githubbachelor, 2016-10-03:
I build a complete new Sound System with all files checked in a Studio and new recorded With digital Voice. The voices 1010 t0 133 are not official designed in the Horus Firmware. But some Voices in the System are without a number and I am waiting for the direction from these sounds to their special number.

The zip file "de" you have to unzip into an emty folder "DE" !! Than you have any new sounds in the folder de you maybe can use and some fun. The "SYSTEM" Folder is completely.
[de.zip](https://github.com/opentx/opentx/files/505326/de.zip)

Githubbachelor, 2016-10-03:
The new Soundsystem belongs to 2.2.NB358

Githubbachelor, 2016-10-03:
How to setup the Multimodul with the internal Software from Horus OpenTX. How to connect that Modul? What about the Inputs from Modul PE0 and PE1

3djc, 2016-10-03:
This is not the place for questions, or the advertise your  sound pack, It is an issue tracker. Please use rc forums for things like that.

closed 2016-10-03 by 3djc