opentx/opentx #3873: Soundfiles missing 2.2NB358

Githubbachelor, 2016-10-03:
System Soundfiles are not available from 0110 to 0134

3djc, 2016-10-03:
And what is your problem exactly ? (what do you expect the radio to say and what is happening) ?

Githubbachelor, 2016-10-03:
Other answer: I am missing the Soundfiles from 0110-0134 in Sounds/De/System on card. in Horus 2.2NB358.
These soundfiles are available in the Taranis Open TX. Are the files from Taranis Open TX 0110-0134 are the same? Can I use them in the 2.2NB358 Version?

3djc, 2016-10-03:
You did not answer my question : what did you expect the radio to say and what happened. Sounds numbers have been reworked in 2.2

Githubbachelor, 2016-10-03:
These soundfiles in the System are not available in the System under their numbers.... IF THEY ARE RIGHT???? If they are right I can correct all. And I prepared all files new in my Sound Studio. All other Sound files I have resampled again for the 2.2NB358. That I have done. But these files are in the list and not there!

"0110;;Volt",
"0111;;Ampére",
"0112;;Meter pro sekunde",
"0113;;",
"0114;;kilometer pro stunde",
"0115;;Meter",
"0116;;Grad",
"0117;;Prozent",
"0118;;Milliampere",
"0119;;Milliampere pro stunde",
"0120;;Watt",
"0121;;db",
"0122;;Fuesse",
"0123;;Knoten",
"0124;;Uhr",
"0125;;Minuten",
"0126;;Secunden",
"0127;;R P M",
"0128;;g",
"0129;;Timer",
"0130;;Timer",
"0131;;Sendung",
"0132;;Empfang",
"0133;;A1",
"0134;;A2",

3djc, 2016-10-03:
The units are no more listed as number. Depending on the language, you will have volt0 up to volt5 for volts for example

Githubbachelor, 2016-10-03:
why did you do that ? That is not logical? I have seen that there are any new files for example Volt0, or founce0, degree0. But why did you not give them numbers as before? Are the number from 0110-0134 reserved for other things?

3djc, 2016-10-03:
Closing this as it does not sound (no pun intended) as a firmware/companion issue

closed 2016-10-03 by 3djc

Githubbachelor, 2016-10-03:
I am working every day with Horus OpenTX now and I am testing all. What do you think we got the possibility to work over Companion, to save and load the models and got them on the sreen for edit?

Githubbachelor, 2016-10-03:
I build long time ago before Taranis and co the first Flysky TX with 3 different TX Modules internaly.