opentx/opentx #3874: Model Names can not be renamed

Githubbachelor, 2016-10-03:
Model Names can not be renamed. If you do System crashed.

3djc, 2016-10-03:
Once again, not a single detail ....

What radio type ?
What steps did you exactly follow so that we can reproduce the issue

Githubbachelor, 2016-10-03:
If you want to rename for example Model1 into the name you gave that Model1 (example Corsair) the Horus will shut down. 2.2 NB358

3djc, 2016-10-03:
I cannot duplicate i can rename without issues

Githubbachelor, 2016-10-03:
I tryed it now. I can rename the Model3.bin into Test3. But after if you go to exit, the name Model3.bin is back again and the new Name Test3 is nowhere to find.

3djc, 2016-10-03:
That's not how model are renamed. You go in the first line of model setup and the first line allow you to change the name of the model. The filename are used for system purposes only

Githubbachelor, 2016-10-03:
OK. How many models can I build in the Horus? As much as I want? I have more than 100 models at home. In the Taranis I can load the models back from SD Card. How to do in the Horus?

3djc, 2016-10-03:
I believe 100 (0 to 99)

Githubbachelor, 2016-10-03:
At my Taranis with Firmware 2.0.20 I can store all my models on Card. And I  can change my models and reload my models If I want. So on card can be millions of models! What can I do in Horus 2.2NB358

RealTadango, 2016-10-04:
Add the required models.bin to the models.txt in the RADIO folder. That way you can add and removed them from the models list without deleting them.

3djc, 2016-10-04:
I would not advise end users to do that unless they where somewhat IT literate

closed 2016-10-05 by 3djc