opentx/opentx #3875: tranis telemetry not working

tjinbristol, 2016-10-04:
Frsky fvlss and current sensors not working, 
x8r will not bind in d16 1-8 mode the keeps the telemetry from working will only bind in d8 mode

3djc, 2016-10-04:
Need more details, it works fine here

projectkk2glider, 2016-10-04:
Possibly the same issues as #3867

tjinbristol, 2016-10-04:
yes the same issue just with tranis, the cells or current not show up in discovery, version 2.9 in the radio, Tj

On 10/4/2016 02:53, Damjan Adamic wrote:

Possibly the same issues as #3867<https://github.com/opentx/opentx/issues/3867>

schwabe, 2016-10-04:
Binding in D8 mode will most likely disable smart port sensors like flvss. Are both receiver and internal module same firmware? (I.e. both EU LBT or both FCC?)

closed 2016-10-05 by 3djc

tjinbristol, 2016-10-05:
reopen this , the problem is the receiver will not bind in d16 mode 1-8, PLEASE tell me hoe to fix it, thank you, Tj

3djc, 2016-10-05:
This is not for end user support, but for bug tracking (which yours is not). You can seek help in opentx forum : http://openrcforums.com/forum or rcgroups :http://www.rcgroups.com/forums/showthread.php?t=2255974

But in all likelyhood, schwabe already gave you a more than serious hint