opentx/opentx #3879: Horus Crash when trying to setSERIAL PORT S Port Mirror

Githubbachelor, 2016-10-07:
Horus shut down into Emergency Mode, when trying to set in Main Menu, Site Hardware and Names ,SERIAL PORT - choose S Port Mirror. After second shut down. Is that OK???

kilrah, 2016-10-07:
The low level functions needed for it to work are still missing at this point.

kilrah, 2016-10-23:
Same happens when setting telemetry type to "FrSky D (cable)", that obviously doesn't apply to the Horus and should not be selectable.

Githubbachelor, 2016-10-24:
Hallo Kilrah, do you have an idea when these missing points are available? And when is the NB 358 an official firmware/Software. Also most important would be to complete Companion for loading and transmitting models, preparing all on desktop?
By the way, can you leave me an privat email with your privat contact? I have a very good idea, never on the market to talk about. You can not believe how important for future!!!

Greetings Michael from Germany

kilrah, 2016-10-24:
No time estimation on anything, official release will be once all important known bugs are fixed and companion support for Horus is added.

You can find my address in my Github profile

bsongis, 2016-11-18:
It should be impossible to select "FrSky D (cable)" on Horus

bsongis, 2016-11-18:
USART3 is not started in this mode (missing call to USART_Cmd(SERIAL_USART, ENABLE))
I wonder if it works on Taranis, we have the same code

bsongis, 2016-11-18:
Confirmed it's also broken on Taranis. I will fix it tomorrow

commit a083342c (2016-11-19): Fixes #3879
closed 2016-11-19 by bsongis