opentx/opentx #3882: Lua Widget VALUE not usable. Strange behaviour when editing the value

RealTadango, 2016-10-07:
Both in Simulator and Horus editing the value does not work properly. When changing the number it lockes at a random max and min with. Simple lua to test:

`local options = {
	{ "Val", VALUE, 0 }
}

function create(zone, options)
	local context = { zone=zone, options=options }
	return context
end

function update(context, options)
	context.options = options
end

function refresh(context)
	lcd.drawText(context.zone.x, context.zone.y, "Val: " .. context.options.Val)
end

return { name="Test", options=options, create=create, update=update, refresh=refresh }
`

projectkk2glider, 2016-10-07:
You failed to provide all members of the options array. Each option must have 5 fields:
 * name
 * type
 * defult value
 * min value
 * max value

RealTadango, 2016-10-07:
I found the struct and understood the option..... but if i provide a min and max value (limit) the widget does not even show up. If the widget was added this message appears on startup: Lua disabled! Nothing in the debug output of the simulator though....

So this does not work at all:
local options = {
	{ "Val", VALUE, 0, -100, 100 }
}

projectkk2glider, 2016-10-07:
>Nothing in the debug output of the simulator though

Meaning that it works in simulator? Or something else.

RealTadango, 2016-10-07:
The Lua disabled message is from the sim. The Widget is not loaded and cannot be selected.

projectkk2glider, 2016-10-07:
There is an error message in simulator output, just difficult to find :grimacing: 
```
PANIC: unprotected error in call to Lua API (bad argument #-1 (table expected, got number))
```

projectkk2glider, 2016-10-07:
Fixed

RealTadango, 2016-10-07:
I was getting close to the fix myself :) Maybe next time i can fix it myself. Thanks for the quick update ;)

projectkk2glider, 2016-10-07:
Man, if you would tell sooner I would gladly leave it to you...

RealTadango, 2016-10-07:
Ha no i am not that var yet but i hope to understand the code better every day. Finding the right files / functions is the main part now and i just found it. Maybe next time you get a solution with the problem :)

projectkk2glider, 2016-10-07:
This issue reminded me to create #3885

RealTadango, 2016-10-07:
I think something is still wrong :) If you look at function luaGetInputs you see that field 0 is name, 1 is type, 2 is min, 3 is max and 4 is default.... min and max were also set when type is read as 0 (integer) to -100 and 100 so it should have worked normal during edit. 

That is a different order that you specified and that i see in the default widgets?

projectkk2glider, 2016-10-07:
Unfortunately the order of items for  **inputs** and **options**  differs (as you found out). What I also don't like is that there is no default value for any member of an option that is not specified in the script. Lets reopen this...

So basically two issues remain:
 * unification of the order of members for **inputs** and **options** 
 * default values for unspecified members

RealTadango, 2016-10-07:
Tested on Horus and confirmed to be working as expected.

Sadly this was not the bug that caused weird behavior for the widget when adding, i will open a new bug.

commit 88e17404 (2016-10-07): Fixes #3882: Widget option min an max values were not parsed (resulted in Lua disabled)
PR #4391 (2017-02-04): Re #3882: parse theme/widget options depending on option type
This addresses the comment https://github.com/opentx/opentx/issues/3882#issuecomment-252257264 . I decided not to change the order of options.

Closes #3882
closed 2017-02-04 by bsongis