opentx/opentx #3884: Horus OpenTX Special Functions Incremental and Source

tomkfly, 2016-10-07:
I have been trying to assist someone else set up a Jukebox on his Horus, but the only  Function seems to be "Value" , "Increment" and "Source" don't show for selection.

projectkk2glider, 2016-10-07:
Are you trying to adjust (increment/decrement) GVARS? If so, the compile option -DGVARS=YES should be used (not sure if it is in nightly builds).

3djc, 2016-10-07:
Gvar is an ondemand compilation option on companion :)

tomkfly, 2016-10-07:
Thanks Projectkk2glider, the problem now is +=1 is making the GVAR increase 10.
3dcj GVARS are the default. You need to check the " nogvars" box to deselect them.

projectkk2glider, 2016-10-07:
I see the problem that increase never stops as long as condition is true. In my mind it should only do one increase or decrease when the condition becomes true.

projectkk2glider, 2016-10-07:
Even if you use EDGE LS, that switch is ON for 100ms which is enough to do 10 increments instead of one.

tomkfly, 2016-10-07:
It makes more sense to only increase by the value selected once per event. Inthe setup I am using the condition is true for 0.1 seconds, but that was put in just so the switch could be seen when monitoring in Companion Sim. I will test it with the duration at zero and see if that fixes it.
Thanks

projectkk2glider, 2016-10-07:
Exactly, this needs to be fixed. Can you type in a new issue for this, this one is too contaminated.

tomkfly, 2016-10-07:
Done

closed 2016-10-07 by tomkfly