opentx/opentx #3886: Lua widget on Horus crashes the system when "selecting"

RealTadango, 2016-10-07:
I have a widget that causes the widget selector to behave strange and crash. When the widget is selected (by scrolling through the widgets) the selection of the zone jumps to another zone. The system crashes (emergancy mode) if i select the widget or wait. This does not happen on the simulator.

I think the reason is drawing out of bounds (of the zone). If i remove the output and add the widget it works when i add the output back in. I will test some more.

3djc, 2016-10-07:
Yes,  widgets are running code that interact with the radio. And yes,  they are powerful to ensure they can do great things,  but you can do bad things with those.  It is the responsibility of the widget developer to ensure this doesn't happen

closed 2016-10-07 by 3djc

projectkk2glider, 2016-10-07:
Still not matter what kind of script user puts together it **must not crash the radio**. 

The Lua API in 2.1 was pretty stable, but changes in 2.2 introduced so many new things that this kind of behavior (crashes) are expected, but each one should be hunted down and fixed.

RealTadango, 2016-10-07:
local options = {
	{ "LineColor", COLOR, WHITE }
	, { "Source", SOURCE, 1 }
	, { "Interval", VALUE, 100, 10, 1000 } 
	, { "Min", VALUE, -100, -1000, 1000 }
	, { "AutoMin", 2, 0 }
	, { "Max", VALUE, 100, -1000, 1000 }
	, { "AutoMax", 2, 0 }
}

To many options? 4 options work correctly