opentx/opentx #3895: Can't adjust GVARS with trim sw.

bigbobed, 2016-10-09:
OpenTX 2.2.0 RC6(N358) cant adjust GVARS with trim switches.

SF2 SB- Adjust GV1 Throttle Trim on

kilrah, 2016-10-09:
Works fine here. Please check and supply more details if confirmed, including useful things like what radio you're using.

bigbobed, 2016-10-09:
Sorry. Radio is Horus X12S production.

bigbobed, 2016-10-09:
Don't know what other detail I can provide. Simple SF line. Works fine on a X9E running 2.1.

kilrah, 2016-10-09:
Well... Create a new model, describe exactly what you configure and how you're determining it doesn't work.

Again, all fine here on Horus.

bigbobed, 2016-10-09:
Not sure what to say. No matter what I did, I couldn't make it work. Loaded FrOs to try to do something similar in that firmware. Posted the issue and worked on FrOs. Following your request I reloaded OpenTx and created a new model and as you suggested, it worked. Reloaded the original model and it worked also. Must be a poltergeist in the house.

closed 2016-10-09 by bigbobed

bigbobed, 2016-10-09:
On further review, I had N536 loaded by mistake when it failed. When it worked I had correctly loaded N538.