opentx/opentx #3898: Lua widgets color not fully working like c widgets?

RealTadango, 2016-10-11:
I try to create some color in widgets and am refering to /src/lua/api_lcd.cpp for the options. I currently am facing 2 issues:
 - Opacity not possible? I cannot use OPACITY(5) for example as used in the c widgets. Is OPACITY possible in lua?
 - How can a global color (like MAINVIEW_PANES_COLOR) be used in a widget? I can change it through setColor but i cannot use it like 	lcd.drawFilledRectangle(x, y, width, height, SOLID, MAINVIEW_PANES_COLOR) If this would be possible the theme colors can be used in the widgets.

RealTadango, 2016-10-11:
I stand corrected: Statics like MAINVIEW_PANES_COLOR do work, i was confused by the OPACITY missing for those colors.

3djc, 2016-10-12:
No opacity is not available to LUA

RealTadango, 2016-10-12:
It looks like a simple method declaration for the lua lcd part. I will build it myself as soon as i have the companion build working.

RealTadango, 2020-01-07:
More lua options still very welcome :)

closed 2021-10-12 by stale[bot]