opentx/opentx #3901: Frsky D16 default channel number + Latency in Menu

schwabe, 2016-10-12:
Taranis uses D16 with 8 channels as default and Horus uses D16 with 16 channel as default. Most user will probably leave as on  the default number.

The difference between these two is that D16 with more than 9ch has a higher latency of 18ch than D16 with <= 8ch (9ms). (Verified with a logic analyzer and sbus ouput of a XSR and the reversed engeeneered code in the DIY Multimodule also confirms this).

DSM2/X has the same problem (11ms <= 7ch, 22ms/11ms depending on channel for 8-12ch and 22ms > 9ch, weird 88ms/44ms/22ms for >12ch) but Spektrum is more honest about this.

Should we add an indication to the UI to specify the framerate that is actually used?

3djc, 2016-10-13:
I personnaly think that D16 is the right default choice given SBUS becoming more mainstream and the release of new receivers with gyros which require 16 chans.

I'll have a look at the UI to see if we can display it without too much troubles

3djc, 2016-10-13:
I need more details. Do you get larger lantency if you go past chanel number 8, or if the total (last-first) goes beyond 8 ?

kilrah, 2016-10-13:
It's the number of channels sent. 
If you send 1-8 channels the same 8-channel packet is sent every 9ms. If sending more than 8, then packets for 1-8 and 9-16 are sent alternatively. 

I'd go with the usual default of 8.

3djc, 2016-10-13:
I have created a PR for the display of latency.

As fas as default, I can understand that when sbus and newert receivers where not part of the landscape, there was little point in defaulting to 16. But now, I think more people will have an issue of S6R/S8R not working than the ones bothered by a 8ms latency increase

kilrah, 2016-10-13:
X receivers and SBUS work perfectly fine with 8 channels.

A few might forget to increase the number of channels when they need them, but the fix is usually found pretty quickly.

3djc, 2016-10-13:
schwabe,

We are discussing the pro and cons of each options, and I came to check on Horus. It does not seem to me that it is using 16 chan by default, is it happening for you ?

schwabe, 2016-10-13:
@3djc I don't have a physical one but the simulator does it for the external module but not for internal. So internal is fine.

3djc, 2016-10-14:
I have commited the latency display, Closing

closed 2016-10-14 by 3djc

projectkk2glider, 2016-10-14:
@3djc but is the default for external module fixed?

3djc, 2016-10-14:
The first issue is to agree on what it should be (which we haven't completely, no ?)

projectkk2glider, 2016-10-14:
I think we agreed on 16 channels.

3djc, 2016-10-14:
ok, making all 16 in a separate PR

schwabe, 2016-10-14:
@projectkk2glider that is the more intrusive change. All to 8 only changes external on 9x and Horus, while all 16 changes it for every internal module

projectkk2glider, 2016-10-14:
After a long discussion we agreed that using 16 channels default in D16 mode is the best solution.