opentx/opentx #3904: OpenTX 2.19: 3 Point Switch in Stick not supported?

dklueh79, 2016-10-13:
I don't know, if it is an hardware or software problem - I think it's OpenTX.
The Problem is, that a  3-pos-switch (from http://rctechnik.de/Knueppelschalter-Spektrum/10003.html) doesn't work correctly. The switch works well, it has been measured. But if I activate it in OpenTX under switches, the main-screen shows an urgly screen, the picture attached. Surely it's the Version for Taranis X9E

The 3-pos switch only works for one postion. Do you have any solutions?

[image]

[image]

3djc, 2016-10-13:
The screen is normal when you start adding more switches on X9E. (you probably declared it to be connected on SR. The radio does not care (or know) where the stick is , or what it does look like

dklueh79, 2016-10-13:
I configured the stick in main menue. But the strange is, that the 3-pos switch only supports middle off, down -100% and up postion nothing. I've check the second delivered stick- it behave the same way.
It's connected to sw-l

[image]

3djc, 2016-10-13:
Strange your first screen show R, not L. You need to compare with existing switch to ensure the are the same

dklueh79, 2016-10-13:
Yes, really strange. so I switched to sw-q. Installed as 3-pos switch. But the same. It works in upper pos with 100%, middle zero and down is nothing. Could it be a mistake in wiring? but i think the wiring is the same like the switches build in.

[image]

kilrah, 2016-10-13:
They a "smaller" letter, e.g. I or J... I know there was an issue with some higher ones that can't be fixed in 2.1, will have to move to 2.2 for them to work.

dklueh79, 2016-10-13:
You mean, if I use a slot that exists like SA (sw-a) it should work?

kilrah, 2016-10-13:
I said SI or SJ...

dklueh79, 2016-10-13:
Ok, the strange is, that if I connect the Stick-switch to SB where the 9E has an original 3-pos, too, the new stick only works as 2-pos....I think it could be a problem of the switches, not OpenTX....if I connect the original SB to sw-i it works as an 3-pos....

kilrah, 2016-10-13:
Then 

> The switch works well, it has been measured.

Isn't correct :)

closed 2016-10-13 by kilrah