opentx/opentx #3910: S6R LUA-scripts follows simplified/wrong grammar to make coding easier. It should be helpful to the one using it.

fraternl, 2016-10-15:
I played with the S6R LUA script a while ago and didn't finish any calibration because the text to it is in chinglish. I opened the LUA-script to see if I could improve it, but found it used 1 sentence for all and only changes a word.

`local calibrationPositions = { "up", "down", "forward", "back", "left", "right" }`

I think the correct way of doing it is using proper English sentences like "place right side up", "turn upside down", "roll" if necessary (I wouldn't know, because I stopped the wizard and it's now gathering dust).

30 years ago one could have an excuse that it's impossible to do on a small display with only a few pixels. IMHO there is no excuse for it now. These tools should be self explanatory and not contain shortcuts to ease programming.

If it needs a manual it's broken

3djc, 2016-10-15:
Fixed in https://github.com/opentx/opentx/pull/3906

closed 2016-10-15 by 3djc

fraternl, 2016-10-15:
That's very good to hear...
I will see if I can check that out....