opentx/opentx #3911: Special Function to control the widgets

fraternl, 2016-10-15:
On my electric planes I use a widget that displays the image of the plane and 2 timers. I think that looks nice. When I'm flying that image of the model isn't really necessary anymore.
I was looking for a special function to change the screen, but it seems it doesn't exist.

Otherwise I would have coupled my throttle safety to it and show a different screen during the actual flight.
Could we have that? Please don't answer with "LUA"  ;-)

RealTadango, 2016-10-18:
Something like a special function: Set screen (1-xx, Next, Previous) would be nice.

kilrah, 2016-10-18:
Would need very careful handling of cases where the user specifies a non-existing screen or one is deleted after being set up.
There is currently no interface for the mixer system to control the UI.

closed 2016-11-27 by fraternl