opentx/opentx #3912: Deleting file in SD manager updates the list wrongly

kilrah, 2016-10-15:
[image]

Deleting "P51Mus.png" results in this:

[image]

The correct file has actually been deleted, it's a display problem

projectkk2glider, 2016-10-22:
Can't reproduce on `simu` (Horus and Taranis) with ae47b1a4d1b2dc5470e427d010aa844f4e365bdf

kilrah, 2016-10-22:
It seems to depend where the cursor is in the list. You might need to scroll down a bit , then go back up above the point that causes scrolling if you get what I mean

projectkk2glider, 2016-10-22:
Confirmed, the item must not be on first page (scroll needed), Taranis is not affected, there the whole list is refreshed after the delete and cursor is moved to the first line.

projectkk2glider, 2016-10-31:
I chosen the simplest solution, the cursor is moved to the top (like Taranis).

commit 258b8adf (2016-10-31): Fixes #3912: Deleting file in SD manager updates the list wrongly (Horus)
closed 2016-10-31 by projectkk2glider