opentx/opentx #3923: Batch-Uploading files in Windows to the SD Card freezes Taranis

ChrisPrefect, 2016-10-17:
I have a windows batch-script that uploads many firmware files to the taranis. This worked flawlessly up to the latest version of the firmware.

If I now start the batch file which just copies files to the usb mounted sd card drive, the copy process locks up on the first file and the taranis freezes up. I have to remove the battery to get it working again.

Something on the usb mass storage driver seems to be buggy?

Uploading settings in the companion works as expected.

Thanks!

schwabe, 2016-10-17:
what version are "latest version" and "previous version"? (i.e. upgrade from 2.1.8 to 2.2rc6)

ChrisPrefect, 2016-10-17:
there was only one firmware update within the last two weeks that companion app downloaded I guess.

I could investigate further if you are not already aware of this bug.

projectkk2glider, 2016-10-17:
@ChrisPrefect we need to know the exact version that is giving you trouble. And if possible the previous version that worked for you.

ChrisPrefect, 2016-10-17:
current version ist 2.1.9, all previous versions worked.

I update about 10 new taranis per week, so I am always up to date and it has nothing to do with the taranis itself.

It happens in mass storage mode. This is the batch command that gets stuck now since the last version:

copy "C:\...\9XR\Taranis SD Card\FIRMWARES" E:\FIRMWARES

thanks! Chris

schwabe, 2016-10-17:
@ChrisPrefect which mass storage mode. The one that comes which the bootloader or the one when the Taranis is actually showing the OpenTX UI.

projectkk2glider, 2016-10-17:
@ChrisPrefect thanks for data. It would help if you can take out the SD card from the radio and check it with the USB card reader. Just to verify that you do not have the issue with the SD card itself.

projectkk2glider, 2016-10-17:
I think you experienced this issue #3840. It is already fixed and will be in 2.1.10. In the meantime please use the Bootloader.

closed 2016-10-17 by projectkk2glider