opentx/opentx #3927: Open TX Horus 2.2 ON352 - Flight mode trim assignments not working

clivemartin, 2016-10-17:
Example, selecting trim from another flight mode + own trim as offset. This function does not appear to work.

kilrah, 2016-10-17:
I have a model set up that uses this config and have no issue.

You should update your firmware to the current build before reporting issues, especially as it's still pre-release there are many bugs that get solved everytime, you're 4 builds behind.

clivemartin, 2016-10-17:
Sorry, will update to 358

clivemartin, 2016-10-17:
Still have an issue despite updating to 358.  If cross trims are selected in Input menu (elev trim on Thr trim) then the fault as detailed above holds.

kilrah, 2016-10-17:
Please post a non-working setup (ideally the .bin file of that model on the SD card).

projectkk2glider, 2016-10-30:
Could this #170 be related?

projectkk2glider, 2017-01-14:
Duplicate of #170

closed 2017-01-14 by projectkk2glider