opentx/opentx #3934: Feature Request Companion: Copy - Paste to All Profiles

RipperDrone, 2016-10-18:
Many of us have multiple RC models defined on our Taranis. Sometimes a special function or logical swith has to be added and is supposed to be applied to ALL models. This has to be accomplished by a straining sequence of copy-paste operations 1 by 1 in 2.1.9. It would be of great help if after copying the new item there was a "Paste to All Models" menu item to facilitate these operations.

kilrah, 2016-10-18:
This will certainly not happen since there's no way to guarantee that the same LS / SF number is not already used on any other existing model or conflicts in any other way. References could also not be made automatically, so if anything you might have more risk to forget parts of the setup.

Might want to try using the global functions (and global LSs when they're added in 2.2).

closed 2016-10-18 by kilrah

fraternl, 2016-10-18:
@kilrah  You're closing these requests everytime. There is a genuine need to have a partial paste of a model. I think it can be done

All could be overwritten except model name/image receiver settings and outputs
This would make the setup the same as the source model, but keep the settings of the target model.
The result would be a setup that's as correct as the original.

It would be very very handy if you have a complex glider setup and several gliders and you want to make a change which needs to be redistributed to the other models.

kilrah, 2016-10-19:
I haven't closed a request in months, but this one is just unreasonable. 
It is NOT asking about what you're describing at all (which would be more feasible, and -surprise- there is an issue for it and it's still open), but about pasting a single SF ot LS line to all models. Please read before whining.

RipperDrone, 2016-10-19:
@kilrah Your reasoning makes perfect sense to me - I was aware it will introduce some risk of making mistakes / overwriting existing SF/LS. On top, I was unaware that the SF/LS functionality is going to be extended that much in 2.2 (sorry, I'm no expert in GitHub and did not browse all the dev branches and comments yet). So you are sying that the full SF/LS functionality will be implemented as global (model spanning) functions? If so, I'm the happiest man on Earth. 

Background: I've implemented a quite sophisticated 2-stage low voltage battery alarm (using edge / stcky functions as per Oscar Liangs guide). Since I'm flying 4S on all my models, the implementation would be identical throughout all models. However, in current OpenTX 2.1.9 there are no global LSs available therefore I needed to code / copy-paste this trigger function individually for all my 6 (!) quads. This was big pain and I thought there should be easier ways. Copy - Paste for all models was the first thought that came to mind, but if there's another way using global SF/LS, I'm perfectly ok with it. Will be even safer, as you have rightly pointed out.

3djc, 2016-10-19:
But it is worth noting that telemetry data a not accessible from global SF/LS

RipperDrone, 2016-10-19:
arrrrggghhhh - this mean I cannot apply my low bat warning as a global function ;-(