opentx/opentx #3939: Enhancement request for LUA: require function

pau33, 2016-10-20:
It looks, as if there is no 'require'-function in the LUA-implementation of opentx (Taranis Plus, 2.1.8).
Is this correct, and if so, are there plans to bring this function to LUA.
It would help a lot to structure the LUA-code.
Best Regards, Paul

projectkk2glider, 2016-10-28:
There are several possibilities http://stackoverflow.com/questions/2932172/how-to-create-include-files-in-lua-language:
 * dofile 
 * loadfile
 * require

`dofile` is working out of the box. Test script:
```lua
local function run(event)
  print("test")
  dofile("foo.lua")
  return 1
end

return { run=run }
```

Script `foo.lua`:
```lua
print "hello from foo.lua"
```

Result:
```
test
findTrueFileName(foo.lua)
	found: foo.lua
f_open(foo.lua, 1) = 0x7faf8c01bd10 (FIL 0x7faf9c436058)
f_close(0x7faf8c01bd10) (FIL:0x7faf9c436058)
hello from foo.lua
Script finished with status 1

```

projectkk2glider, 2016-10-28:
`loadfile` also works:
```lua
local function run(event)
  -- loadfile test
  f = loadfile("bar.lua")
  f()
  barFunc()
  return 1
end

return { run=run }
```

Script `bar.lua`:
```lua
function barFunc (x)
	print("Hello from bar")
end
```
Output:
```
findTrueFileName(bar.lua)
	found: bar.lua
f_open(bar.lua, 1) = 0x7fffdc013ae0 (FIL 0x7fffed584048)
f_close(0x7fffdc013ae0) (FIL:0x7fffed584048)
Hello from bar
Script finished with status 1
```

projectkk2glider, 2016-10-28:
The `require` is currently disabled, in fact the whole `package` library is disabled. Enabling it would take up some RAM, which we do not want on the Taranis. 

@pau33 could you use `loadfile` and `dofile` instead?

projectkk2glider, 2016-10-28:
@lshems any comment on this? I know your scripts use some kind of dynamic loading.

pau33, 2016-10-28:
Thank you for the response,
There are some small differences between require an dofile (for ex. no check if loaded twice).
"require" looks to be the actual way in Lua.
But in this content I can use dofile as well.

loadstring and loadfile are afaik deprecated in lua 5.3 it's now only load. I assume, the lua version in OpenTx is an older one. It is more a kind of inline execution, but works quite well in some scopes.

Form my point of view, we can close the point.

projectkk2glider, 2016-10-28:
OpenTX uses Lua 5.2.2.

closed 2016-10-28 by projectkk2glider

lshems, 2016-10-28:
What I often use is 

`f=f or loadfile("filetoload")`

then the file isn't loaded again if it was already loaded.

Also, by passing a table, you can "fetch" functions by a loadfile:

create a file "library.lua" with :
T=...
T.f1 = function() return 1 end
T.f2 = function() return 2 end
return T

now, you can load all these functions in one go:
library={}
t=loadfile("library.lua")(t) 

and t.f1() will return 1
t.f2() will return 2

There are some small differences with "require" or other similar things, and I first also was missing some library functions, but the way I do it now is actually more convienent. It allows you to delete the functions after use, freeing up memory again.

hope this helps, and we need every bit of memory we can for running the scripts, and preferably not block it in including not mandatory  stuff by default.