opentx/opentx #3944: Companion simulator has Hold X and Hold Y, needs Hold SF.

Miami-Mike, 2016-10-21:
I programmed my Taranis to enable certain switch functions while I hold switch SF, but I can't test it with Companion because there's no way to hold SF like you can with the sticks. My only workaround was to temporarily change it to another switch.

kilrah, 2016-10-21:
I guess you mean SH...

But what about setting all other things as they should be first, and actuate SH last?

Miami-Mike, 2016-10-21:
What I'm working on is a special "Set Failsafe Trims" flight mode for adjusting trims to the best-flying failsafe positions while I'm actually in flight. In order to properly test this in Companion I need to secure SH↓ and then actuate all of the trims to make sure they do what they're meant to do without affecting the other flight modes. 

It  works great, by the way.

kilrah, 2016-10-22:
Can be taken into account whenever we finally do that full simulator revamp.

pindeslandes, 2016-10-27:
Please, see #1627

mpaperno, 2017-01-15:
Duplicate of #1627, closing.

closed 2017-01-15 by mpaperno