opentx/opentx #3946: No option for Haptic on Taranis X9D Non-Plus

hactiv8, 2016-10-22:
Ive ticked the box in companion 2.20N359 and i still can't get the haptic option to appear on my taranis I've even compiled from source and still not getting this as an option is there something I'm doing wrong or a bug somewhere?

kilrah, 2016-10-23:
You must re-download firmware and flash it after ticking the box.

hactiv8, 2016-10-23:
Im flashing it with opentx-x9d-haptic-noheli-lua-multimodule-sqt5font-en-2.2.0N359.bin so its downloaded it with the haptic option but it's still not working

schwabe, 2016-10-23:
I think he is right, that is a bug in the Cmake script.

For X9D we do
```  set(HAPTIC NO)```

instead of using an option that could be changed

kilrah, 2016-10-23:
Yep, correct

projectkk2glider, 2016-10-24:
Fixed in #3949

commit 8c870312 (2016-10-23): [X9D] Allow toggling Haptic option for Taranis X9D  (closes #3946)
commit 73a7027e (2016-10-23): [X9D] Allow toggling Haptic option for Taranis X9D  (closes #3946) (#3949)
closed 2016-10-24 by projectkk2glider