opentx/opentx #3954: linker support not included on Horus

raleighcopter, 2016-10-26:
can support for the linker be added for the Horus?

schwabe, 2016-10-26:
What do you mean with "linker"?

raleighcopter, 2016-10-26:
https://alofthobbies.com/linker.html

takes the sbus or cppm stream from a receiver (d4r-ii or x4r-sb and pipes it in through the module bay as a trainer port to wirelessly link to another transmitter as a master or slave.

kilrah, 2016-10-26:
The hardware of the Horus is different and that pin is unfortunately not a serial port like it is on the Taranis, so at least SBUS is likely to be out unless someone feels like decoding serial from timer captures.

raleighcopter, 2016-10-26:
If cppm will work, it would be very nice to have.

agibson2, 2017-03-05:
If that is possible for CCPM to work, Is that something time consuming to implement or would most of the code come from the previous implementation on the Taranis?  I would love to be able to use the Wireless Trainer from alofthobbies in CCPM mode.

3djc, 2017-03-05:
We understand the need, but as mentioned earlier, it will be a time consuming exercise, so it is unlikely for 2.0

robustini, 2019-09-11:
With the SBUS there is little to do, but with the CPPM you can do it, but not via software:

https://www.rcgroups.com/forums/showpost.php?p=42694565&postcount=16448

raphaelcoeffic, 2019-09-11:
see #5518

robustini, 2019-09-12:
> 
> 
> see #5518

Excellent alternative @raphaelcoeffic , but perhaps it is more complex to weld those two resistances rather than inject the CPPM signal where the radio expects to have it natively.
However good to know.

brainbubblersbest, 2020-05-03:
Here is your x12s ixjt mainboard
[image]
[image]

closed 2021-01-30 by stale[bot]