opentx/opentx #3956: lua: lcd.drawNumber() standard-alignment changed from "RIGHT" to "LEFT"

strgaltdel, 2016-10-26:
oTx RC7 (&prevous revisions) changed luascript standard alignment for numbers from RIGHT to LEFT
This will cause some redesign of most scripts which uses the "lcd.drawNumber" function

simple test with following statements:

	lcd.drawNumber(50,25,123)
	lcd.drawText(lcd.getLastPos(),25,"  lcd.getLastPos() : ", SMLSIZE)
	lcd.drawNumber(lcd.getLastPos(),25, lcd.getLastPos(),LEFT + SMLSIZE)

2.1.8

[image]

2.2 rc7

[image]

tested on Taranis X9e,
Companion screenshot represent firmware behavior

issue was discussed some weeks ago on rocketchat

3djc, 2016-10-26:
Yes,  all text functions,  including drawn number,  now default left.  We realize it is a change and some scripts are going to need rework (adding RIGHT for example).

strgaltdel, 2016-10-26:
OK, so if this is final state i can start to adopt existings scripts, no problem

Thanks
Udo

3djc, 2016-10-26:
Thx for your understanding.  We did investigate reverting that,  but that change was made so early in 2.2 dev that there was a lot of implications.

closed 2016-10-26 by 3djc