opentx/opentx #3957: OpenTX 2.2: Breaking changes in voicepacks, but no CSV file to reference changed filenames

lazd, 2016-10-26:
I worked on editing a [voicepack for OpenTX 2.0](https://miawinther.bandcamp.com/album/mia-taranis-voicepack), and with OpenTX 2.1, we had to update some of the filenames and record new sounds. Luckily, there was a [super sweet .csv file of the OpenTX 2.1 sounds](http://voices-21.open-tx.org/opentx-taranisplus/en/english-taranis.csv) for us to cross reference against [the .csv from OpenTX 2.0 version](http://voices-20.open-tx.org/opentx-taranisplus/en/english-taranis.csv) and, though it took some time to manually sort through all of the tracks, separate the new tracks and record them, then rename dozens of sounds that had their filenames changed, we were able to upgrade the pack.

However, we can't find this file for OpenTX 2.2, so we don't have a way to update the voicepack for OpenTX 2.2.

Is there anything like this available?

Additionally, it would be nice if there was an upgrade guide, with a list of the new tracks and their filenames and a mapping for old tracks that are renamed, or even a shellscript that renamed existing tracks for you. This is quite painful for people creating OpenTX voicepacks, so some support in this regard would be greatly appreciated.

Thanks!

kilrah, 2016-10-26:
It's in the published SD card archive in the SYSTEM folder it refers to.

There might be a guide made but not before official release since things might still change.

closed 2016-10-26 by kilrah

3djc, 2016-10-26:
In addition, the script to generate the nightly sound pack is here, if that can help : https://github.com/opentx/opentx/blob/next/radio/util/tts_en.py

lazd, 2016-10-26:
Thanks for the quick reply, @kilrah and @3djc! I found the CSV and should have everything I need. I will wait for the 2.2.0 proper release before upgrading the sound pack in case things change.

For others looking for this file, I downloaded a nightly SD card archive from here:
http://downloads.open-tx.org/2.2/nightly/sdcard/

The file is inside located at `SOUNDS/en/SYSTEM/en-US-taranis.csv`.