opentx/opentx #3961: Horus Production Unit - O/S Model menu interface hangs on Pg down (random)

Webbsolution, 2016-10-28:
On random attempts to access the Model menu immediately followed by pge down  - the interface hangs... 

One other pilot has experienced this as well. The link is in tact and sounds function... switches function... but the software menu interface hangs. I hard power off ( hold power button for 10 seconds) is required.

projectkk2glider, 2016-10-30:
Can't seem to be able to replicate on beta Horus with self debug build (CLI and DEBUG and bunch of other stuff enabled). Will keep an eye out for this.

Webbsolution, 2016-10-30:
Try binding...then model then page down is fast succession...

projectkk2glider, 2016-10-30:
Binded to D8 receiver on a model with 16 sensors - can't reproduce. Strange. I might try with the official N359 firmware next. @Webbsolution what firmware options did you check in the Companion?

projectkk2glider, 2017-01-14:
Closing for now, reopen if it is reproducible.

closed 2017-01-14 by projectkk2glider