opentx/opentx #3962: Lua scripts have a 6 char limit in 2.2?

floaledm, 2016-10-28:
Lua scripts placed in the /SCRIPTS/ folder of the SD card and launched through the CUSTOM SCRIPTS and DISPLAY menus can only have 6 characters max in their filename (extension excluded). If the file names are longer, the files are not listed.

Any idea what is causing this? I'd be happy to make a PR on it, but I'd appreciate a little headstart  by letting me know where I should look...

3djc, 2016-10-28:
Yes,  it is everywhere filename are used (sounds,  images)  and Lua scripts as well.

floaledm, 2016-10-28:
Thanks for your confirmation.

Files (including lua scripts listed and (and executed) from the SD CARD menu can be more than 6 chars (e.g., SR6_Calibrate.lua), but for some reason it's limited to 6 elsewhere...

In previous versions (e.g., 2.1), the names can be at least 8 chars. What changed in 2.2? Can it be fixed? I'd be happy to take a stab at it with some guidance...

projectkk2glider, 2016-10-28:
This was intentional, to get back some space that is now used by 32 additional special functions.

closed 2016-10-28 by projectkk2glider

floaledm, 2016-10-28:
Thanks Damjan for the quick response! Sounds reasonable. Let's leave it at that then.

floaledm, 2016-10-28:
Looks like in 2.2, the number of inputs to lua custom scripts is also limited to 6, whereas more inputs were possible before. Could someone please confirm whether that is by design?