opentx/opentx #3964: luaSportTelemetryPop troubles when using DIY passthrough

floaledm, 2016-10-29:
A while ago, I had tested 75e1bead3f121c7bb6515846dff51e346ed81923 before it got merged and that was working fine: for a packet with a Data ID 0x5001, calling `id,value = telemetryPop()` from a lua script would give me id=1 and the value was good.

Now, using `physicalId, primId, dataId, value = sportTelemetryPop()` I get primId=16, dataID=0x5001 and the value is good for a while (about 10 seconds after transmitting that packet repeatedly).

But after that, I run into a problem as I permanently receive weird data that looks like it's shifted by some number of bytes: primId=1, dataId is wrong (I explain right below) and value=452984832 no matter what value I actually transmit.

So when the weird behavior happens:
If I transmit a value of 1, I get dataID=336 which in binary is 01 01010000.
If I transmit a value of 2, I get dataID=592 which in binary is 10 01010000.
If I transmit a value of 3, I get dataID=848 which in binary is 11 01010000.
So, it looks to me as if the data is shifted, but it only happens after I've been transmitting for a little while. Restarting the Taranis clears the problem, but it comes back again about 10 seconds later.

Also, when this happens, if I transmit on Data ID 0x5002, I get primId=2.

Please do not hesitate to ask followups, and I can certainly do some tests, even recompile the firmware with modifications...

DeepnineRC, 2016-10-30:
Same on Horus 359.

If you send your DIY id's slower it works. 10 normal telemetry values then one DIY id (10,1,10,1...)
I use this as work around at the moment.

projectkk2glider, 2016-10-30:
Looks like you are not reading fast enough from the telemetry FIFO and when it gets full it will drop a byte. This would then result in the shift you observe.

We need to add a check if we have enough free space in the FIFO before putting in the new data.

floaledm, 2016-10-30:
I am transmitting on the FrSky almost as much as the link can take on DIY IDs, so roughly I estimate it's something like 30 packets per second. I think your assessment is correct that when the FIFO gets full, things get shifted one byte, because if I call sportTelemetryPop() more often (more than once per loop of the main run() function), I don't get the problem. Your PR is still necessary in cases where the FIFO gets full so thank you for working on that...

Just a suggestion but I wonder if it wouldn't be best to drop the oldest packet when the FIFO gets full, rather than dropping the new one. I suppose the problem is that crossfire packets are not a fixed length so it might be tricky to do that...

kilrah, 2016-10-31:
You should always pop in a loop until empty!

floaledm, 2016-11-01:
This issue can be closed since #3968 addresses it...

PR #3968 (2016-11-02): Fixes #3964: Added free space check before putting new data into Lua …
…Telemetry Fifo

Closes #3964

Completely untested!
commit a332f7a8 (2016-11-02): Fixes #3964: Added free space check before putting new data into Lua … (#3968)

* Fixes #3964: Added free space check before putting new data into Lua Telemetry Fifo

* Copy-paste error

* Compilation fix

* Another compilation fix
closed 2016-11-01 by floaledm