opentx/opentx #3966: Audio going to max volume on startup

JapieG, 2016-10-29:
On my Horus oTX 359

Situation:
A model is created for a race quad. 6 position selects flight modes. Every flight mode has a SF to play track speaking out the flight mode. I.E. Position3 is Angle Mode.

A global function is made to assign RS to regulate system audio volume.

Radio is turned of, 6 position know into position3 (angle mode)

When i turn on the radio i hear 'Welcome to OpenTX' on average volume.
Then after some switch warnings the model file loads. I hear on max volume 'ANGle mode'. Where the capital letters stand for the max volume. I assume that during the playback of the soundfile the global function for volume is loaded and the volume returns to the selected value.

Is it possible to load the global functions before the model file loads? Or maybe easier to set default system volume to the same level als 'Welcome to OpenTX' message? I can not remember having this issue in versions before RC7? 

Not a high priority issue i think, but just wanted to report it ;-)

Thanks for your great work!

kilrah, 2016-10-29:
Already fixed for next nightly.

closed 2016-10-29 by kilrah

projectkk2glider, 2016-10-30:
Fixed in 0eba5c4c9c5a587992f43dcc2a528bc0b805aacd