opentx/opentx #3979: Telemetry stoped working / iXJT and MultiModule

SoulPictures, 2016-11-01:
I updated about a week ago to **N359** and subsequently updated my Multimodule too.
Telemetry worked fine using my **Multi Module with [v1.1.5](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.1.5)** and i got RSSI feedback from my TinyWhoop (DSMX, 6ch, 22ms). Sunday i noticed that my telemetry using the internal Model was not working. (iXJT @1.2.16 with X4R-SB connected to a Motolab Tempest running Betaflight 3.0.1). i flew a few lipos without telemetry, switched quads, still no telemetry. Having a short break i turned of my **Horus** Some minutes later I turned it back on with telemetry working fine for the rest of the day. (i only flew IXJT based models that day) Today i noticed that the Telemetry is not working with my multi module too. I Checked if it is working with my XSR/X4R models, but i didn't work there either. I left a model using the internal IXJT modul selected and switched of my transmitter. After turning it on again telemetry worked with all my different quads (which have different models, i am not using one model for all of them). I tried switching over to the TinyWhoop using the Multimodule and the telemetry there was not working. If i switch back to my iXJT Models it stays broken until i reboot my transmitter. 

Telemetry worked a few days with N359 and the multi module so i don't have a direct clue what is causing the issue. Do you have any ideas/suggestions what i should check to trace the error?

kilrah, 2016-11-02:
Looks like the serial port might not be reconfigured correctly when switching from multi to FrSky or vice versa, requiring a reboot for the right settings to be applied.

SoulPictures, 2016-11-02:
What was kind of weird today was that the telemetry using the multi module started working again. The Horus was booted using the MultiModule with a Model using receiver 01, afterwards switched to 00. I will try again tomorrow but yet i don't see any structure in this. I'll keep you updated

schwabe, 2016-11-04:
What protocols are your multi model and the internal model? Both D16?

SoulPictures, 2016-11-04:
My multi module is primarily used for DSM X 22ms with 7/6 channels.
The internal is only set up in D16 and there using 8ch only

On the MultiModule i excluded FrSky X/D/V Protocols to fit others while flashing but left all telemetry options active.

projectkk2glider, 2017-01-14:
@schwabe can this be closed?

commit 8e23c1d6 (2016-12-18): Implement a custom telemetry for the multi module

The custom telemetry allows the module to tell the telemetry protocol to OpenTX and also allows signaling the status of the Module (e.g. invalid protocol) to OpenTX

This should also close #3979.
commit c062a382 (2016-12-18): Implement a custom telemetry for the multi module

The custom telemetry allows the module to tell the telemetry protocol to OpenTX and also allows signaling the status of the Module (e.g. invalid protocol) to OpenTX

This should also close #3979.
commit f4dde658 (2016-12-18): Implement a custom telemetry for the multi module

The custom telemetry allows the module to tell the telemetry protocol to OpenTX and also allows signaling the status of the Module (e.g. invalid protocol) to OpenTX

This should also close #3979.
PR #4151 (2016-12-23): Schwabe/multi telemetry
The custom telemetry allows the module to tell the telemetry protocol to OpenTX and also allows signaling the status of the Module (e.g. invalid protocol) to OpenTX

This should also close #3979.
closed 2017-01-14 by schwabe