opentx/opentx #3987: Request: Adjust brightness per model / trhough lua / ??

RealTadango, 2016-11-03:
I am adjusting the brightness every time i use the Horus indoor (50%) or outdoor (90-100%). With so many features there must be a better solution. It would like an option to set the (on) brightness per model (indoor vs outdoor models) or set it through lua (for even more intelligent options). The perfect way for me would be through a light sensor, but that might be not an option without a modification (not for everyone that is). 

So, is there something possible in the model data or a setting adjustable through lua?

schwabe, 2016-11-03:
I think handling that like volume should be fine (e.g. new special function for that). Haven't looked if adding SF without eeprom is possibly.

3djc, 2016-11-03:
Beside the potential eeprom issue, I think we need to set a minimum so that you do not end up with a black screen .. forever :D

RealTadango, 2016-11-03:
Well... that might be a good idea ;) setting SF ON brightness 0 would make most people unhappy :)

kilrah, 2016-11-03:
Minimum is handled in the driver, it won't accept anything below 5

projectkk2glider, 2017-01-14:
Related to #3408 where the original intent of Backlight SF was to change color.

nmaggioni, 2019-03-12:
Pinging this back to life, having per-model brightness override would be very handy: when flying wings having the screen always on (even on reduced brightness) lets me peek at telemetry widgets anytime, but with FPV quads or simulators there's no need to waste precious battery life.

RealTadango, 2019-09-09:
This would still be usefull for those indoor or night flying sessions ;) Using a pot or per model setting is still very welcome.

RealTadango, 2020-03-09:
Still a welcome feature :)

HermanKruisman, 2020-09-05:
Still a welcome feature :)

3djc, 2020-09-05:
If only ! Oh wait it has been added already !

closed 2020-09-05 by 3djc

RealTadango, 2020-09-05:
Ow.... Awesome :)

RealTadango, 2020-09-06:
> If only ! Oh wait it has been added already ! You have a special function to adjust brightness now

Will be in 2.3.10? When was this added?

3djc, 2020-09-06:
https://github.com/opentx/opentx/pull/7680