opentx/opentx #3995: Telemetry log messed up

Allerhopp, 2016-11-05:
2.2.0RC7@X9E
Telemetry works fine, but the log is messed up. GPS and LSW possibly in a wrong format? Source is an openXsensor, which worked flawlessly with prior OpenTX versions. No changes made since.
[new  1.zip](https://github.com/opentx/opentx/files/573114/new.1.zip)

Allerhopp, 2016-11-14:
When i deleted the sensors and discovered them again, GPS did work as expected, but the LSW field still has the wrong format. Is there a recommendation, to update the sensors, when updating to 2.2?
[image]

projectkk2glider, 2016-11-14:
Nice, a while ago some other user had the same problem with GPS, same wrong logging contents for location. Since 2.2. is not stable yet, there were a couple of internal storage changes that probably caused this. 

What do  you expect to see in LSW field. That is a hex representation of 64 hex number, each bit is one LSW. Decoding in Companion still needs to be implemented.

Allerhopp, 2016-11-14:
Thx, i did not realize, that the LSW field represents the logic switches. It is sometimes hard to know, which findings are helpful and which are not ;)

closed 2016-11-14 by Allerhopp