opentx/opentx #4002: Add support for TX Side mode selection (Telemetry, CH 1-8, 9-16)

RealTadango, 2016-11-08:
FrOS has an option during bind to select the operation mode of the receiver. This would be nice to have in OpenTX of course :) There are 4 modes with the combinations of Telemetry on/off and channel 1-8 of 9-16. This would probably need some info from FrSky for the commands send during bind....

schwabe, 2016-11-08:
We have no information how to communicate that to the IXJT. Sniffing FrOS to IXJT serial protocol with a logic analyser is probably the fastest way to get information at the moment.

kilrah, 2016-12-24:
My contact at FrSky repeatedly insisted it was an S-Port command, but I believe I did find the responsible flags in the PXX stream as expected in below capture for future reference.

Receivers actually don't support that yet, so not putting any priority on it.

[image]

PR #4743 (2017-04-06): Enable set binding mode as a firmware option for frsky radios
Add an option to allow X6R/X8R/XSR mode set at binding time (much like Horus FrTx)

This fixes #4002
closed 2017-04-06 by bsongis