opentx/opentx #4010: [Simulator] main window does not show up on OSX

raphaelcoeffic, 2016-11-09:
This issue has been observed on OSX Yosemite all across the next branch, and is thus probably related to Qt5 in some way.

I looked for anything strange with GDB and could not find anything. The process is not stuck in anything, it's just not showing the main simulator window, that's it.

Any clue/pointer from someone knowledgable on QT would be highly appreciated.

bsongis, 2016-11-19:
Isn't it just a packaging problem?

schwabe, 2016-11-19:
no, unfortenately not. The simulator window even works, you can hear sounds from the simulator but the window is not visible.

raphaelcoeffic, 2016-11-19:
With gdb on, I could not find anything suspicious, but please bear with me that I have close to 0 experience with QT.

bsongis, 2016-11-20:
I have some experience with Qt but none with MacOS. Perhaps we can try to work together and fix it!
On Linux I just launch 'make' then I can test my binary
I wish it's the same on OSX, but I have an error with a Cocoa lib missing

joaquins, 2016-11-21:
I can reproduce the same issue using the latest nightly (opentx-companion-2.2.0N359.dmg) and also compiling the source code myself.

This issue is NOT happening when compiling for HORUS. Launching simulator.app with libopentx-horus-simulator.dylib works as expected (simulator window opened and responsive).

If using the X9D+ build (libopentx-x9d+-simulator.dylib), the simulator opens but no window at all is visible.

raphaelcoeffic, 2016-11-22:
@joaquins that's interesting! I might be able to find the difference between both.

raphaelcoeffic, 2016-11-22:
@bsongis I do run it right on the spot where it has been compiled. Basically, the same as you do on Linux.

joaquins, 2016-11-23:
Ok, this is weird: I tried today again compiling in a different computer, with OSX Yosemite, and the simulator works perfectly with both Taranis and Horus!

I guessed that it's related to the building environment and the QT versions used, but I just copied the binaries (simulator.app & libopentx-x9d+-simulator.dylib) into the first system (OSX Sierra), and the issue is still there: Taranis simulator window NOT showing.

So, something in the taranis UI is not working with OSX Sierra only?
@raphaelcoeffic which version of OSX are you using?

raphaelcoeffic, 2016-11-25:
@joaquins I'm still on Yosemite. I could try your binaries, just to see if it would work here as well.

joaquins, 2016-11-30:
I've been digging a bit more, but still does not make a lot of sense... My Yosemite machine had also QT 5.5.2.1 installed, and maybe this makes the difference, but my cmake command uses 5.7.0 (same as in my Sierra machine).

I kept trying different things, and finally isolated the problem into simulatordialog-taranis.ui. If I rearrange the switches and dials kind of randomly (deleting the current grid layout), the main window works again.

As soon as I try to add a new grid layout or even horizontal layout, the bug is back... so no final solution found yet.

Maybe you can try to build again using my simulatordialog-taranis.ui at: https://github.com/joaquins/opentx/commit/f436953c4945682e1d7d88d047a1bc29c58ed184#diff-91efc4a959319292f8c9f1a6a1ce4a2a

Let me know if that works.

bsongis, 2016-11-30:
Nice catch! It would be really nice if you could isolate the element which brings the problem, perhaps removing switches one after one?

bsongis, 2016-11-30:
I will check what valgrind says now

joaquins, 2016-11-30:
@bsongis the thing is... is does not make much sense. If you see my edited ui, all the elements are still there, just out of the original gridlayout.

I tried lots of combinations, removing one by one, or editing some properties, and nothing is clear. Just QT does not like the layout as is, for some reason. And only in Sierra. And only wth the Taranis ui.... really weird.

@raphaelcoeffic can you confirm if my edited .ui works also in your Mac, just to double check?

raphaelcoeffic, 2016-12-05:
@joaquins sorry for letting you wait so long! I'm trying it right now.

joaquins, 2016-12-05:
@raphaelcoeffic let me now if it works with both.

But, that would only confirm that there is a problem with some Qt versions and that ui... still no idea what the issue really is.

raphaelcoeffic, 2016-12-06:
@joaquins the windows is now displayed in the X9D+ simulator. In case it has any importance, I have QT 5.7.0 installed with Homebrew.

joaquins, 2016-12-06:
Thanks @raphaelcoeffic . Then it's confirmed that the issue is related to QT 5.7.0, not Sierra vs Yosemite.

The only "fix" I found was to move the switches/pots away from the grid layout they are originally, but that is not a proper fix, of course.

schwabe, 2016-12-06:
QT 5.5 vs 5.7.0 makes no difference for OS X 10.9 and X9D+ simulator for me. Both don't work. On Sierra both work.

joaquins, 2016-12-06:
@schwabe can you check the output of "brew info qt5" in both systems? Specially in your Sierra, if it's working.

This is the output I got in my System with Yosemite (OSX 10.10.5).

`joaquins$ brew info qt5
qt5: stable 5.7.0 (bottled), HEAD [keg-only]
Version 5 of the Qt framework
https://www.qt.io/
/usr/local/Cellar/qt5/5.5.1_2 (7,246 files, 292.9M)
  Poured from bottle on 2015-11-16 at 23:29:16
/usr/local/Cellar/qt5/5.7.0 (8,218 files, 327.6M)
  Poured from bottle on 2016-11-22 at 11:16:38`

Screenshot of the taranis x9d+ working on latest nightly, with such system above.

[image]

schwabe, 2016-12-06:
@joaquins the qt version are installed via the offical qt installer instead of homebrew. Homebrew does not work if you want to compile software for older OS X versions. 

I am building the release on a 10.11 machine. The setup is described here:
https://github.com/opentx/opentx/wiki/OpenTX-2.2-OSX-Build-Instructions#building-a-release

joaquins, 2016-12-06:
@schwabe ok, so right now the only valid combination is OSX 10.11 or greater + QT installed using the installer instead of homebrew?

I will try with the QT installer in my OSX Sierra as you suggest.

The fact that my OSX 10.10.5 with homebrew QT was working fine, was just misleading me then? :(

joaquins, 2016-12-08:
Ok, finally found something that makes sense: it's related to the size of the Taranis window, and the laptop screen resolution.

For my Sierra laptop, screen size 13" (2560x1600), neither QT 5.7 or 5.5 worked. BUT, as soon as I resize the height of the Taranis simulator (editing the simulatordialog-taranis.ui), it works.

I just reduced the minimun height of both ledtStick and rightStick from 245 to 200, and the simulator window now shows up as expected.

schwabe, 2016-12-08:
I just tested that with my 10.9 VM. I can confirm that changing the screen resolution makes the OpenTX simulator work.

schwabe, 2016-12-08:
Issue #3135 is kind of a duplicate of this. Different symptons but same underlying problem

joaquins, 2016-12-24:
This is fixed with https://github.com/opentx/opentx/pull/4154

@raphaelcoeffic please try the latest code in NEXT branch. The window size has been reduced and it shows as expected.

schwabe, 2016-12-24:
On my mac os 10.9 vm everything worked

raphaelcoeffic, 2017-01-06:
Sorry for the delay, I just tested on my side as well, and it all works fine. Should I close this issue?

kilrah, 2017-01-06:
Yep, was validated on our side, but jsut waiting on your feedback

closed 2017-01-06 by kilrah