opentx/opentx #4014: Port to Turnigy Evolution?

maymayday90, 2016-11-11:
Just wondering if there are any plans to port OpenTX to the Turnigy Evolution radio? Next big upcoming radio that would love some OpenTX love.

kilrah, 2016-11-11:
Need to see what the hardware is, its cousins had way too little memory.

maymayday90, 2016-11-11:
The poor dim cousins... Never stood a chance.

Pics of the main board (and the gimbals):
[image]
[image]

Hardware is basically the same as the i6s, good documentation on hardware here:
http://fishpepper.de/2016/09/15/openground-part-1-components-pinout/

Not sure how easy it would be to make a "slim" version of OpenTX? This radio is primarily a multirotor radio (3 switches, one pot), so wouldn't need any heli or other crazy mixing. Most of it is done in the flight controller anyways.

Lemme know if you need any other info about the radio and I can get what I can.

kilrah, 2016-11-11:
Meh, 64kB part again.

The AR9X version takes 186kB. We can shave off some by removing audio and SD card, but probably never that much without copious amounts of work.
64kB is about what OpenTX takes with none of the fancy options on a 9x, and ARM code notoriously needs more memory than AVR for the same thing so chances are very slim.

Wonder what that flex with a chip on it is on the left.

kilrah, 2016-11-11:
That said, given you say 
> This radio is primarily a multirotor radio (3 switches, one pot), so wouldn't need any heli or other crazy mixing.

I'd imagine the stock firmware does enough, wonder what you'd want to gain by using OpenTX.

projectkk2glider, 2016-11-11:
Actually  STM32F072VB should have 64kB or RAM and 128kB of FLASH

[image]

I am interested in its cousin FS-i6s for DLG usage (small, light). But it has some drawbacks like:
 * no sd card
 * no physical trim switches
 * not compatible with FrSky out of the box

maymayday90, 2016-11-11:
Flex chip goes to either the LCD or the touchscreen, can't tell without pulling the whole thing out.

Reasons for wanting OpenTX on this little beast:
-PPM output (either internal solder point or over USB interface?) for external modules (including support for DSMX, Frsky and Crossfire CRSF)
-Better control over what the switches do (reading telemetry data, throttle up, etc)
-Better throttle timer settings
-Better model memory handling
-Expo

kilrah, 2016-11-11:
Oops, confused the B with a 8... Better then.
Yep, touchscreen makes sense. LCD is the other large ribbon in the middle.

> -Better control over what the switches do (reading telemetry data

No audio to read it with...

maymayday90, 2016-11-11:
Ah poop, that's true. Just beeps of various tones. Possible to mod with a better speaker?

Biggest thing is connecting external modules. Is there anyway to get PPM out of the feed to the internal RF module? Or is it a serial com that'd need a converter? Or somewhere else it could be located?

kilrah, 2016-11-11:
There's no chance threre's PPM between the processor and module, the module is likely jsut the RF chip the processor would talk to using SPI or similar, and handle all the protocol details.

projectkk2glider, 2016-11-11:
Lots of info here http://fishpepper.de/projects/openground/

The interface with RF module is SPI, see http://fishpepper.de/2016/09/27/openground-part-9-cc2500-some-ui-work/

maymayday90, 2016-11-11:
Ya, saw fishpepper's code and blog, but he isn't actively developing it as far as I can tell. Last time he updated was when I talked to him a while back.

Days like today I wish I knew how to code something like this. Is the code the same between the two architectures (STM stuff?) Maybe if I ever get a chance to hack it up I could try.

Sad face. Just wanted my Crossfire on here and I'd be a happy man. Maybe I'll just run the output from the receiver to the Crossfire Tx, use it as a relay. Means groundstation jazz but I guess that works.

jacobbloy, 2017-02-15:
Support the porting of OpenTx to the turnigy evolution and I6s.
http://www.gofundme.com/3dudryg﻿

schwabe, 2017-02-15:
To get a **rough** idea how touch/close we are I started from a Sky9X build and delted all object files that related to functions that are not used/could be removed. Sky9X firmware is 194kB with default options.

Removing brings it down to the region of 128 kB but considering that ARM is probably growing instead of getting smaller, will be a tough a target to keep. OpenTX is probably to large to really fit on a 128 kB chip nowadays. Just look at the state of the AVR targets in OpenTX. They are missing a lot of features that ARM has because they don't have the flash space. Also adding even more ifdefs to keep space constraints is not a thing I am looking forward to.

IIrc DeviationTX has the same problem it has trouble to keep the flash size of its smallest remote. And they are already doing tricks like loading the current rf protocol in memory from external flash to keep flash usage down.
```diff
--- original.txt	2017-02-15 10:50:55.000000000 +0100
+++ removed.txt	2017-02-15 10:49:06.000000000 +0100
@@ -1,5 +1,4 @@
    text	   data	    bss	    dec	    hex	filename
-   9584	      4	   4778	  14366	   381e	./radio/src/CMakeFiles/firmware.dir/audio_arm.cpp.o
     840	      0	      0	    840	    348	./radio/src/CMakeFiles/firmware.dir/crc16.cpp.o
    1704	      0	     65	   1769	    6e9	./radio/src/CMakeFiles/firmware.dir/curves.cpp.o
       0	      0	      0	      0	      0	./radio/src/CMakeFiles/firmware.dir/debug.cpp.o
@@ -42,7 +41,6 @@
    2640	      0	      1	   2641	    a51	./radio/src/CMakeFiles/firmware.dir/gui/common/stdlcd/model_inputs.cpp.o
    2852	      0	      0	   2852	    b24	./radio/src/CMakeFiles/firmware.dir/gui/common/stdlcd/model_mixes.cpp.o
     444	      0	     63	    507	    1fb	./radio/src/CMakeFiles/firmware.dir/gui/common/stdlcd/popups.cpp.o
-   2520	      0	      0	   2520	    9d8	./radio/src/CMakeFiles/firmware.dir/gui/common/stdlcd/radio_sdmanager.cpp.o
     562	      0	    198	    760	    2f8	./radio/src/CMakeFiles/firmware.dir/gui/common/stdlcd/view_text.cpp.o
     822	      0	      0	    822	    336	./radio/src/CMakeFiles/firmware.dir/gui/common/stdlcd/widgets.cpp.o
     116	      0	      0	    116	     74	./radio/src/CMakeFiles/firmware.dir/gui/common/widgets.cpp.o
@@ -51,7 +49,6 @@
     310	      4	     17	    331	    14b	./radio/src/CMakeFiles/firmware.dir/haptic.cpp.o
       4	      0	      0	      4	      4	./radio/src/CMakeFiles/firmware.dir/io/frsky_sport.cpp.o
     464	      0	     50	    514	    202	./radio/src/CMakeFiles/firmware.dir/keys.cpp.o
-  26608	      0	      0	  26608	   67f0	./radio/src/CMakeFiles/firmware.dir/loadboot.cpp.o
    1663	      0	    579	   2242	    8c2	./radio/src/CMakeFiles/firmware.dir/logs.cpp.o
     678	      2	     12	    692	    2b4	./radio/src/CMakeFiles/firmware.dir/main_arm.cpp.o
      40	      0	      0	     40	     28	./radio/src/CMakeFiles/firmware.dir/maths.cpp.o
@@ -61,8 +58,6 @@
     244	      0	      0	    244	     f4	./radio/src/CMakeFiles/firmware.dir/pulses/ppm_arm.cpp.o
     388	      6	    243	    637	    27d	./radio/src/CMakeFiles/firmware.dir/pulses/pulses_arm.cpp.o
    1624	      0	      2	   1626	    65a	./radio/src/CMakeFiles/firmware.dir/pulses/pxx_arm.cpp.o
-   1370	      0	     13	   1383	    567	./radio/src/CMakeFiles/firmware.dir/rtc.cpp.o
-   2468	      0	      7	   2475	    9ab	./radio/src/CMakeFiles/firmware.dir/sdcard.cpp.o
      81	      0	      0	     81	     51	./radio/src/CMakeFiles/firmware.dir/stamp.cpp.o
     362	      0	    720	   1082	    43a	./radio/src/CMakeFiles/firmware.dir/storage/eeprom_common.cpp.o
    4106	      0	      0	   4106	   100a	./radio/src/CMakeFiles/firmware.dir/storage/eeprom_conversions.cpp.o
@@ -88,27 +83,9 @@
     810	      0	      0	    810	    32a	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/pulses_driver.cpp.o
      52	      0	      0	     52	     34	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/pwr_driver.cpp.o
     152	      0	      4	    156	     9c	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/rotenc_driver.cpp.o
-     74	      0	      8	     82	     52	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/rtc_driver.cpp.o
-     80	      0	      0	     80	     50	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/sdcard_driver.cpp.o
     268	      4	    520	    792	    318	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/serial2_driver.cpp.o
     420	      0	    134	    554	    22a	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/telemetry_driver.cpp.o
     312	      0	      3	    315	    13b	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/trainer_driver.cpp.o
-    130	      0	      0	    130	     82	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/usb/common/core/USBConfigurationDescriptor.c.o
-     30	      0	      0	     30	     1e	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/usb/common/core/USBEndpointDescriptor.c.o
-     22	      0	      0	     22	     16	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/usb/common/core/USBFeatureRequest.c.o
-     14	      0	      0	     14	      e	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/usb/common/core/USBGenericDescriptor.c.o
-     46	      0	      0	     46	     2e	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/usb/common/core/USBGenericRequest.c.o
-     22	      0	      0	     22	     16	./radio/src/CMakeFiles/firmware.dir/targets/sky9x/usb/common/core/USBGetDescriptorRequest.c.o
[... cut ...]
-    493	      0	      0	    493	    1ed	./radio/src/CMakeFiles/firmware.dir/translations/tts_pt.cpp.o
-    375	      0	      0	    375	    177	./radio/src/CMakeFiles/firmware.dir/translations/tts_se.cpp.o
-    532	      0	      0	    532	    214	./radio/src/CMakeFiles/firmware.dir/translations/tts_sk.cpp.o
    7067	      0	      0	   7067	   1b9b	./radio/src/CMakeFiles/firmware.dir/translations.cpp.o
     364	      0	      0	    364	    16c	./radio/src/CMakeFiles/firmware.dir/vario.cpp.o
- 199794	    165	  41580	 241539	  3af83	(TOTALS)
+ 130332	     45	  36433	 166810	  28b9a	(TOTALS)
```

kilrah, 2017-02-15:
The evo has no audio nor SD card, removing that will reduce size significantly

jacobbloy, 2017-02-15:
The next big plan would be to replace the internal pcb, reverse engineer it to have audio and sdcard. With an stm32f205, I would do it if I could get my hands on a copy of the taranis schematic. It would make porting 100% quicker.

kilrah, 2017-02-15:
I've had a quick look around, it seems the Evo is unobtainium right now and some places suggest its future may be uncertain... 

Will be interesting to see what happens.

jacobbloy, 2017-02-15:
It's been such a big seller. Maybe a v2 is in the works.

schwabe, 2017-02-15:
@jacobbloy the taranis schematic is property of frsky. You need to contact Frsky directly to get that schematic under a NDA. The hal abstraction layer of OpenTX tells you what pins are connected where.

I am not sure what reason you have to want OpenTX on the Evolution but using an ARUNi as a replacement mainboard might also be an option.

jacobbloy, 2017-02-15:
I know every one here is opentx, but if it is going to be to hard to port to opentx, would it be better to just add functions to the openground firmware.
Arne I have tried to add you on FB messenger and emailed you a few times about paying you to add multi module support to openground.

schwabe, 2017-02-15:
I genraly don't add people on Facebook I don't know. And I haven't gotten any mail from you about openground. What address did you use? Btw. joining us on the chat will probably a good way to discuss things like this.

jacobbloy, 2017-02-15:
I resent it to your openvpn app support email.

ameeuw, 2017-04-29:
Did anybody find out which pins the LEDs for the gimbals are on? Would love to have them light up on my Evolution when I use OpenGround.

yellobello, 2017-11-05:
Sorry for digging this up again, but since sales for the evo and i6s still going strong it kind of still is an issue.. :-)
I have noticed the 24c128 i2s eeprom on the board and am wondering why not using this chip as firmware storage? Why is it on the board in the first place? As a buffer for loading the fw when updating? 
I have no clue if the serial roms are adressable as firmware storage from the arm´s view.. so if somebody capable of this stuff could comment on it?
If this would be possible, I furthe would have no problems with soldering an 24c256 or 512 in .. its just 8 legs of reasonable pitch.
best regards

schwabe, 2017-11-05:
Might be possible but is probably a real headache to use such a slow i2c eeprom as firmware storage. Also might be to slow as storage

kilrah, 2017-11-05:
Not an option AFAIK. Would have to read the whole serial flash into RAM and execute it from there, but there isn't enough of that either....

yellobello, 2017-11-07:
Hm.. would there be an pin-compatible MCU to swap?  
TQFP100s with this pitch are honnestly not that hard to solder. If you know how it´s a job of 5 minutes.
From a quick check there´s a F3 with 512kb flash that is ALMOST pin compatible... but then.. is it really worth it?
I honestly like the design of the i6s and the evo... all other radios in the 50-200$ range either look like a relic of the 80s or like a piece of plastic refuse. Maybe ARUni is the only way out of this dilemma... oh wait.. it´s sold out...

schwabe, 2017-11-07:
@yellobello For most users soldering a TQFP100 is out of the question. Even I would not do want to solder that.

So that would not be many to use that. And then again someone has port and maintain that port.

closed 2019-08-19 by kilrah

ponnex, 2019-10-25:
[image]
I have questions. Does the internal module of the evo is a transceiver? and if it is are they the same transceiver of the stock rx(FS-IA6C)?

Upon research, PPM out is not possible to get from the board as the MCU and Internal module talk via SPI then to internal module via flysky protocol to the receiver. However the receiver does have PPM out. I'm thinking maybe if the way these transceiver talk is by encoding(from evo) and decoding(receiver then output it to ibus, sbus, ppm) the data from SPI, so by removing the transceiver from the FS-IA6C and wiring it to the transceiver of the evo, we could get ppm out from the FS-IA6C board?

kilrah, 2019-10-25:
No, they won't be talking the same things.